MCUdude/MiniCore #5: cannot reburn bootloader of an Atmega328

Protonerd, 2016-09-06:
Dear MCUdude,

Your board library was recommended to me by users on the Arduino Forum:, related to this issue:
http://forum.arduino.cc/index.php?topic=422239.msg2908127#msg2908127

However I cannot make it work for an Atmega328. Background: usually I use for my boards the 328P, but at the early stage of my project I ordered some 328 which came pre-burned with bootloader (Mini). Because my project needs more code space and the Mini has ~2kB less than the Nano and even lesser as the Uno, I wanted to re-burn the 328 with a new bootloader. Easier I though than exchanging the boards, which would be quite a pain. But it does not want to work...

MCUdude, 2016-09-06:
Now find out what this error really is, you'll need to turn on verbose upload. Open Arduino preferences and enable this. Then try to burn the bootloader again.
[image]

Protonerd, 2016-09-06:
Hi MCUdude,

Thanks a lot for the quick feedback, now I turned on verbose and I get the following:

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Users\MunkaGep\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\1.0.1/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.05s

avrdude: Device signature = 0x1e9514
avrdude: Expected signature for ATmega328P is 1E 95 0F
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

Error while burning bootloader.

MCUdude, 2016-09-06:
It's not strange at all! You have an ATmega328 (non p) that you want to program, but you've selected the ATmega328p. Choose "328/328A", and it should bootload just fine 😃 

[image]

Protonerd, 2016-09-06:
Oh, sorry, I overlooked that! Actually I went the brute force way and changed the signature in the file C:\Users\MunkaGep\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\1.0.1\avrdude.conf, that way it worked as well...but by far not so elegant. Anyway, now I can revert that change as well. I'm really grateful for your promt help!

closed 2016-09-06 by Protonerd

MCUdude, 2016-09-06:
Great to hear!

commit 075403c8 (2019-05-30): Squashed 'avr/bootloaders/optiboot_flash/' changes from 95e2866e..25040bec

25040bec Add 460800 bootloader option
7e913a19 Add CAN32/64/128 URL
4c3c04cf Fix AT90CAN32/64/128 issue The bootloader simply did not work
156c4efd Rename from *eeprom_support to *BIGBOOT This namechange makes sence since BIGBOOT includes other features as well. Flash page copy for instance
924ecda3 avr-libc bug workaround MCUSR is flagged as poisoned in iom128.h. This causes problems for us...
f576a720 Add flash page copy functionality Closes #5
b4e3861c Initial commit
9335dd92 Script improvements + add ATmega325/645  [... cut]