MCUdude/MiniCore #12: A6/A7 mapping issue

ChrisBall, 2017-01-09:
Analog reading "A6" and "A7" on the ATMEGA328P TQFP returns results from the wrong analog ports. Inspection shows these are mapped to 23 and 24, but I get the results I'm expecting from 20 and 21.

MCUdude, 2017-01-09:
I'm not able to check this out right now, but I guess you could fix this by setting [ A6 = 6 and A7 = 7](https://github.com/MCUdude/MiniCore/blob/master/avr/variants/standard/pins_arduino.h#L116-L117). Would you try it and report back?

ChrisBall, 2017-01-09:
This works!

closed 2017-01-09 by ChrisBall

MCUdude, 2017-01-09:
Thanks for reporting! I'll fix it as soon as possible :)

Quick question though; how did you find this repo? Was it through Google, or the Arduino forum? 😃

ChrisBall, 2017-01-09:
I think I must have come here via google, but I honestly can't recall for sure. I'd been developing some projects using TQFP atmegas and I believe I purchased an Atmega328 by mistake (not P) and wasn't able to program it until I found MiniCore. Being able to set the clock frequency/source has also turned out to be really useful :)

sim7755, 2017-05-29:
I Just spent hours trying to  figure out the A6 and A7  at first i thought it was the board , it turns out arduino must have been using an older version than i had thought. it is really great to use though and i apreciate the work you have done it gives much more freedom with my small custom pcbs for me i couldnt fit a crystal on the board and needed more inputs