MCUdude/MiniCore #13: Serial and Wire not working together

beicnet, 2017-05-24:
Hi,

Serial.begin(9600); and Wire.begin(0x27); using ATmega8 with simple Arduino code are not working together, but if I disable the Serial protocol the the Wire protocol works as intended.

Any toughs?

Regards

MCUdude, 2017-05-24:
Weird. I need to test this when i get home. If this turns out to be correct, then there is a bug in the official Arduino source files, which this core depends on. What version of the IDE are you using?

beicnet, 2017-05-24:
Tested again to be 100% shore of it:

Not working scenario:
```
void setup() {
 Serial.begin(9600);
 Wire.begin(0x27);
 Wire.onReceive(receiveEvent);
}
```

Working scenario:
```
void setup() {
 //Serial.begin(9600);
 Wire.begin(0x27);
 Wire.onReceive(receiveEvent);
}
```

I'm using <b>Arduino IDE 1.8.2</b> version.

<b>UPDATE:</b> I found out that the Serial protocol blocks the whole Wire protocol completely (the whole I2C Slave device freezes).

Compiled I2C Scanner and cannot find my I2C Slave device on 0x27 address.

```
Scanning...
No I2C devices found
```

But, as I said, if I remove the Serial protocol, then everything is working fine again!

Regards

beicnet, 2017-05-24:
But, if I rearrange the code like this, then they are working together:

```
void setup() {
 Wire.begin(0x27);
 Wire.onReceive(receiveEvent);
Serial.begin(9600);
}
```
Am I missing something here?

MCUdude, 2017-05-24:
Does it work with the "Arduino NG or older" board option in the IDE?

beicnet, 2017-05-24:
I'm using Arduino Uno as a Master (standard code upload) and a barebone ATmega8 as a Slave on a breadboard with 8MHz external resonator and two 4.7k pull-up resistor programmed by another Arduino Uno with AVR ISP Shield,

MCUdude, 2017-05-24:
ah, ok. That makes it a little difficult suddenly running it at 16 MHz. I believe there's a big in the Wire library, but I can't say for sure until I've done some testing. Whatever the outcome is, I'm afraid there's little I can do about it, because the code that causes this is shipped by the Arduino IDE, not MiniCore.

MCUdude, 2017-05-25:
I've inserted an ATmega8 into my Arduino UNO board to test this. I got one device that acts as an i2c scanner (code below), and the ATmega8 acts as an i2c slave, where I've used your code. This works just fine for me..

```c++
// i2c scanner running at an ATmega2561

/* 
   http://playground.arduino.cc/Main/I2cScanner

   This sketch tests the standard 7-bit addresses
   Devices with higher bit address might not be seen properly.
 */

#include <Wire.h>

void setup()
{
  Wire.begin();

  Serial.begin(9600);
  Serial.println("\nI2C Scanner");
}

void loop()
{
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknow error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);           // wait 5 seconds for next scan
}
```
  
  

```c++
// i2c slave running on ATmega8

#include <Wire.h>

void setup() {
 Serial.begin(9600);
 Wire.begin(0x27);
 Wire.onReceive(receiveEvent);
}

void loop()
{
  
}

void receiveEvent(int val)
{
  Serial.println("Request!");
}

```

beicnet, 2017-05-25:
My setup is: Arduino Uno => ATmega8 => USB2Serial Adapter

MCUdude, 2017-05-25:
Mine too, almost. At least where it matters.

PC terminal 1 -> (UART) -> ATmega2561 <-> (i2c) <-> ATmega8 <- (UART) <- PC terminal 2

I used the ATmega2561 because that was the microcontroller within reach. It's just running a i2c scanner sketch, so it shouldn't affect the ATmega8 at all. You should try replacing the ATmega8 with an ATmega328p, and see if the result changes.

beicnet, 2017-05-26:
After a close inspection, while inserting the Atmega8 every time to the breadboard after programming, it seems that I had a pin contact issues (false alarm), but seemed to real! :)

Sorry for bothering you with it!

Regards

MCUdude, 2017-05-26:
Good to know there issue where caused by a bad connection, and not the code itself. Thanks for reporting 👍

closed 2017-05-26 by MCUdude

beicnet, 2017-05-26:
Yeah, these breadboards are not so reliable, anyway great job you did there with those Cores! :1st_place_medal: