MCUdude/MiniCore #15: SoftwareSerial on ATmega8?

beicnet, 2017-05-30:
Hi,

Can you please implement somehow the "SoftwareSerial" support for the ATmega8 MCU?

Is it possible?

Kind regards,
Viktor

MCUdude, 2017-05-30:
Try using the softwareSerial library found here: [https://github.com/MCUdude/MCUdude_libraries](https://github.com/MCUdude/MCUdude_libraries). It's using regular hrdware interrupts instead of PCINTs, which atmega8 doesn't have. The TX pin can be any pin you want, but the RX pin can only be pin 2 or 3 (INT0 or INT1)

beicnet, 2017-05-30:
Thank for you quick response in a first place! ;)

In that case, can I use both of Serials at the same time, or?

Regards

MCUdude, 2017-05-30:
> In that case, can I use both of Serials at the same time, or?

Yes, that's right. You can use the hardware serial as well as two software serial ports at the same time. That's three in total. Note that the library I linked to is modified and not well tested yet.

beicnet, 2017-05-30:
Oh, great, thank you! :1st_place_medal:

beicnet, 2017-05-30:
and a not related question, I can't find ATtiny85 Core on you Github, am I missing something?

MCUdude, 2017-05-30:
I don't have any Arduino core for ATtiny85. Use [ATTinyCore](https://github.com/SpenceKonde/ATTinyCore)

beicnet, 2017-05-30:
Ohh, sorry, I mixed you with that guy! :(

MCUdude, 2017-05-31:
Did you fix this?

beicnet, 2017-05-31:
Yes, everything seems fine!

Thank you again! ;)

closed 2017-05-31 by MCUdude