MCUdude/MiniCore #17: memset seems to not work?

beicnet, 2017-05-31:
Hi there,

I'm trying a char cleaning like in standard Arduino environment on the ATmega8, but I got error saying:

`invalid conversion from 'char' to 'void*' [-fpermissive]`

Code follows:

```
char pKey;
memset(pKey, 0, sizeof(pKey));
```
Any suggestions?

Regards

MCUdude, 2017-05-31:
Why would you use memset for storing data to a variable?

Have a read over at [cplusplus.com](http://www.cplusplus.com/reference/cstring/memset/). The first argument needs to be void*.

`memset((void*)pKey, 0, sizeof(pKey));`

Just have in mind; MiniCore does not provide _any_ source files, it's just linked to your IDE installation. I cannot do anything to the code to prevent this, at all. If you believe there's a bug, you'll have to report it at the official Arduino repo

closed 2017-05-31 by MCUdude

beicnet, 2017-05-31:
For the standard Arduino UNO usage the code bellow is working:
`memset(pKey, 0, sizeof(pKey));`

and for the Core usage you provided, is working now too:
`memset((void*)pKey, 0, sizeof(pKey));`

Thank you again for your support! ;)