MCUdude/MiniCore #18: Add upload to 250 KBPS version for 8 and 16MHz ?

hallard, 2017-06-19:
Man, 
you saved my life with your working optiboot for Arduino Mini at 8MHz at 3V3 ;-)
Thanks because the original one in optiboot repo does not seems to works, I was unable to upload anything.
By the way would you mind adding a version working at 250000 KBPS at 8Mhz ?
You'll find why my request on the post I wrote 3 years ago here https://hallard.me/ulpnode-bootloader-1/

I'm asking this because compiling optiboot on windows/macOS in really a mess I just tried withiut success

Thanks for your help

MCUdude, 2017-06-19:
Hi Charles! Great to hear you found MiniCore useful. 250000 baud is an ideal speed for an 8 MHz clock, since it got no error. The reason why I haven't included it is because some USB to serial chips and/or drivers struggles with 250000 baud. The Ch340g driver I used with MacOS Sierra just returned jibberish when trying to send or receive content at 250k. It seems to be OK now with the latest drivers for El Capitan. 

I suggest you build the bootloader yourself. Building from source works like a charm on Mac, as long as you have the correct tools installed, and knows how to handle the command line. 

Use [this guide](http://maxembedded.com/2015/06/setting-up-avr-gcc-toolchain-on-linux-and-mac-os-x/) to install the correct tools. I'm compiling with avr-gcc 4.9.3 😊

hallard, 2017-06-21:
Hi,
worked like a charm on my mac after installing and compiling the whole stuff (was long) but now I'm  able to compile back optiboot, thanks
I tested 1M also on nano 8Mhz 3.3V works great
Your arduino menu for boards.txt is also great, may be also have a submenu for upload speed could be usefull ;-)
thanks for your help

closed 2017-08-20 by MCUdude