MCUdude/MiniCore #19: arduinoisp.org Programmer not Working

ChGalle, 2017-06-22:
Hi,

I tried to upload an Arduino pro Mini Bootloader with my Arduino.org-ISP. I get this error message:

avrdude: Can't find programmer id "arduinoisporg"

Trying to programm it via the original Arduino-Boards it works. But not with the MiniCore. 

What am I doing wrong?

Thanks in advance. 

Christoph

per1234, 2017-06-22:
Here's the avrdude.conf patch:
https://raw.githubusercontent.com/arduino/avrdude-build-script/master/avrdude-6.3-patches/31-avrdude-6.3-added_arduinoorgisp.patch

ChGalle, 2017-06-22:
It works!

Thank you very much!

per1234, 2017-06-22:
Glad to help! I don't think the issue should be closed, I wanted to provide you with an immediate solution but hopefully it will be solved for everyone too once MCUdude gets some free time.

ChGalle, 2017-06-22:
Okay. I reopened it. Thank you again.

MCUdude, 2017-06-22:
> Glad to help! I don't think the issue should be closed, I wanted to provide you with an immediate solution but hopefully it will be solved for everyone too once MCUdude gets some free time.

Yep, I'm pretty busy right now, but I'll accept a PR if anyone wants to contribute 🙂

ChGalle, 2017-06-22:
I did a PR for this issue. I hope I did it the right way, cause its the first time doing a PR.

per1234, 2017-07-20:
@ChGalle or @MCUdude, this issue can be closed now. ArduinoISP.org support has now been added fully to all the MCUdude hardware packages.

closed 2017-07-20 by MCUdude