MCUdude/MiniCore #25: platformio

Ahmad110, 2017-09-07:
Hi there 
please add your core on platformio (https://github.com/platformio/platformio-core)
So thanks

sschuhmann, 2018-04-12:
I've switched to this bootloader due to the watchdog problem of the Arduino Pro Mini and my project was build using platformio. Supporting this ide would be awesome!!

JonSilver, 2018-07-23:
Happy to help with setting this up if you're too busy, @MCUdude

JonSilver, 2018-07-24:
@MCUdude if you haven't used PlatformIO for 2 years, you've missed the best bits. And in turn, it's missing your core, because there's little else with proper support for the ATmega328PB. All the options you mentioned are now part of PlatformIO's board definitions and/or project settings (platformio.ini).

MCUdude, 2018-07-24:
Great! I'm kinda busy at the moment, but it would be awesome if you would help me set this up.

So how would the project wizard look like? I believe someone added MighyCore a few years ago, but I'd much rather prefer to only select *MightyCore* in the board menu and choose ATmega1284P, BOD level, clock frequency and LTO in the platform.ini. Is this possible somehow? Would love to see this happened, as it would make PlatformIO much more appealing to me. I have other Arduino cores that could benefit from this as well.

MCUdude, 2019-06-03:
Wish granted! See [PlatformIO.md](https://github.com/MCUdude/MiniCore/blob/master/PlatformIO.md) for details.

closed 2019-06-03 by MCUdude

Donderda, 2019-08-02:
Just one question: When I upload my code via platformio, I have to reburn the bootloader every time before uploading:

1. Burn Bootloader via Arduino IDE
2. Upload code via Platformio, everything works fine.
3. Uploading code again via Platformio ends up throwing this error:
```
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0002
         0x00 != 0x02
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:F6, H:D6, L:FF) 
```
4. Starting at 1.

Am I doing anything wrong? Currently, I disabled the validation of the flashed firmware.

Here's my platformio.ini:
```
[env:MiniCore]
platform = atmelavr
framework = arduino
board = ATMEGA328PB
board_build.f_cpu = 8000000L
build_unflags = -flto

build_flags = 
board_upload.speed = 57600

upload_protocol = usbtiny
upload_port = /dev/cu.usbmodem14101
upload_speed = 57600
upload_flags = -V # << disables validation.

monitor_port = /dev/cu.usbserial-00000000
monitor_flags=
    --raw
    --echo
```

Donderda, 2019-08-02:
Ok, adding the `-V`  upload flag doesn't work either....