MCUdude/MiniCore #29: 4MHz external oscilator - OneWire problem

strange-v, 2017-12-30:
Hello.
I want to use ATmega328P with a super capacitor (voltage range 1.8V - 4.7V). I know that internal oscillator can be a bit unstable, so I want to use external one (4MHz due to min voltage).

I've fixed makeall file, added:
```
make atmega328p AVR_FREQ=4000000L BAUD_RATE=9600 LED=B5 LED_START_FLASHES=2 
make atmega328 AVR_FREQ=4000000L BAUD_RATE=9600 LED=B5 LED_START_FLASHES=2
```
And added 4MHz option to menu.
Then all bootloaders were rebuilt, 4MHz bootloaders were copied do the appropriate directory.

As a result I'm able to burn new 4MHz bootloader and it works, except OneWire communication (i2c - ok).
To test OneWire I've used DS18B20 and oneWireSearch sketch from DallasTemperature library.

Maybe I've missed something that have to be changed?
Thank you.

MCUdude, 2017-12-30:
Hi!
I've never used the 4 MHz clock frequency with the OneWire library, but I'm sure your issue is clock related? Does the oneWireSearch sketch work when running on 1 MHz or 8 MHz internal oscillator?

strange-v, 2017-12-31:
I've just tested 1MHz - no and 8MHz - ok.
After a bit of digging dipper I have to say that this is the false alarm. It seems that default OneWire library requires at least 8 MHz clock frequency.
I've tried [modified one](https://github.com/ginodecock/OneWire4Mhz-ArduinoLibrary) and now I'm able to find my DS18B20.
Thank you for your help!

closed 2017-12-31 by MCUdude