MCUdude/MiniCore #30: Voltage readings not correct?

beicnet, 2018-01-11:
Hi there,

I'm trying to make a barebone ATmega328 voltage meter, but the reading are not correct.

Using a voltage divider resistors 100k and 10k, power supply is 4.96v and I got this readings:
```
4.59
4.64
4.59
4.64
```
Tried all variations of settings, 16MHz external, internal, 8MHz external, internal, burned the bootloader too.

Also pins VCC and AVCC has 100nF capacitors placed close to pins.

But, if I upload the same code to the Arduino UNO board, it's showing me the exact source voltage readings from it:
```
4.96
4.96
4.96
4.96
```
and my code for voltage readings is:
```
#define V0_IN_PIN A0
#define V_REF_INT 5.0
int vIn = 0.00;
float vOut = 0.00;
float vResult = 0.00;
float R1 = 99700.0; //100000.0; // resistance of R1 (100K)
float R2 = 9930.0; //10000.0; // resistance of R2 (10K)

void setup()
{
  Serial.begin(9600);
  delay(1);
  pinMode(V0_IN_PIN, INPUT);
  Serial.println(F("Starting..."));
  delay(1000);
}

void loop()
{
  Serial.println(ReadVoltage(V0_IN_PIN));
  delay(500);
}

float ReadVoltage(int aInpX) {
  vIn = analogRead(aInpX);
  vOut = (vIn * V_REF_INT) / 1023.0;
  vResult = vOut / (R2 / (R1 + R2));
  if (vResult <= 0.09) {
    vResult = 0.0; // Statement to quash undesired reading! 0.09v (Error correction)
  }
  return vResult;
}
```

Can you tell me what I'm doing wrong? Or any help, please? :disappointed: 

Kind regards,
Viktor

MCUdude, 2018-01-11:
To isolate the problem I would start with _just_ printing the values from the analogRead operation. the AVR doesn't handle floats very well

Also I have no idea what your hardware looks like. It's most likely a HW related issue, since MiniCore uses it's source code from the main arduino repo.

beicnet, 2018-01-11:
Same config and same code in ATmega8-16PU gives me constant 4.91v from above mentioned voltage source (4.96v)

[image]

MCUdude, 2018-01-11:
Yes, but what's the ADC value (between 0-1023). Have you measure the voltage *right at* the power pins on the microcontroller using a multimeter?

beicnet, 2018-01-11:
Wait, I have some issues with my power source (wall adapter 12V/1A), now I'm reading 5.66v against mentioned 4.96v ?!? Wuttt??!?!

beicnet, 2018-01-11:
Rebuilt all again onto new smaller breadboard, tested with Battery bank and PC USB and I'm still getting the same results...

[image]

The ADC value 90 to 91

At VCC and AVCC pin the voltage is 4.98v

Thank you for your support! :wink:

MCUdude, 2018-01-11:
I don't think there's a problem at all. If the ADC output the value 90, it's converted to 4.86V. If the value is 91, it's converted to 4.91V. If 92, the voltage is 4.96.

You should also measure the voltage in the right on the ADC pin. Remember that you're also loosing resolution because you use voltage dividers.

beicnet, 2018-01-11:
Or maybe in my multimeter the battery is low or the worst case it's too old and it's getting bad time on time it's like 10-15 years old now... I will replace the battery tomorrow need to buy one.

Yes, I understood all, but how I get exact readings on the legit Arduino Uno board?

beicnet, 2018-01-11:
So, I managed to change the battery in my multimeter, but with same readings mentioned above.

The voltage on the ADC pin is 2.58v - 2.59v

pizzzle, 2018-01-28:
Hey, just wanted to help a bit (not related to this project at all, but I will be trying it as a quick and dirty way to get 1 MHz internal).

Just reiterating the inputs:
VIN = 5V
Vref - 5V
R1 = 100k
R2 = 10k

In your original post ~4.5V is actually correct because remember, the quick math says that one resistor is 10x larger than the other, so it will have 10x more voltage across it, this assumes you're measuring across the 100k (VIN to R2 to R1 to ground). The nearly 5V value is actually wrong.

Something keeps changing in your circuit, so 2.6V is totally wrong. That should result in an ADC count of ~500.

beicnet, 2018-02-22:
I tried everything, can't get any clear readings! :disappointed:

sijskes, 2018-02-23:
There is no guarantee that an ADC will give you a stable value. This is due to noise. On a PCB the noise conditions are quite a bit lower than on a prototyping board. For any quality of ADC you can find a condition where the value changes all the time between 2 steps. The change is higher the more the analogue value reaches the boundary between 2 steps. If you want to display the value measured you need to average the value.

MCUdude, 2018-02-23:
I think this is an issue that should be posted on the Arduino forum, not here on Github. As far as I can see there's nothing wrong with the MiniCore source code, and you'll benefit from a forum post where you'll reach out to many more people than here.

sijskes, 2018-02-23:
@MCUdude indeed. better take this the general ardu forum.

closed 2018-02-23 by MCUdude