MCUdude/MiniCore #33: Enhancement request: Add option to not use bootloader

SpenceKonde, 2018-02-15:
I often field inquiries into using an Atmega microcontroller with the Arduino IDE without using an ISP programmer. I would like to suggest a menu option to choose whether a bootloader is to be used. This would eliminate the issues around the BOOTRST fuse and allow use of the full flash space for user application. It would be great to be able to point these people to your core instead of having to instruct them how to modify boards.txt, and would make MiniCore appropriate for a production environment where a serial bootloader may not be appropriate

MCUdude, 2018-02-15:
Yesah, you're probably right.. 

Not sure if I should include this option inside the *Variants* menu or create a separate *Bootloader* menu. I want to provide lots configurable settings without "bloating" the boards menu. I'd figure it out.

Now I wish there would be a way to change the text "Burn Bootloader" to "Set Fuses" based on a boards menu option.

MCUdude, 2018-02-17:
@SpenceKonde you could/shoud maybe make the bootloader option a sub menu rather than a separate option in the boards menu? It's absolutely doable, and it would make the ATTinyCore boards menu change from the:

  
[image]
  

.. To this:
[image]