MCUdude/MiniCore #41: avrdude fails programming of 328P with USBasp

Teddyz, 2018-03-07:
When selecting low clock frequencies avrdude needs the parameter '-B 5' appended to slow down the flashing. 

1) Is this uncommon? I found out thanks to a post by randy at stackexchange.
2) Is it possible to add it permanently so I don't have to fail a programming and then copy the text and paste it into the cmd-window and add -B 5 at the end?

Thanks!

MCUdude, 2018-03-08:
Hi!
Yes, it's common. I have a USB ASPECTS that has a clock frequency of 384kHz. This does not work with clocks lower than ~1.5MHz.

The -B flag is the clock divider. I often use -B32 when programming really slow microcontrollers.

MiniCore don't have a set of "slow" programmers, but you can install MicroCore if you want a permanent option in the programmers list.

closed 2018-03-09 by MCUdude

Teddyz, 2018-03-09:
Thanks for the answer. I now understand that my workaround solution by adding the -B 5 separately did not solve the whole problem. It only sets the fuses. I suppose that the actual upload of the bootloader is a second step that I never reach since the first step fails in the Arduino IDE.  Can you from the command line below figure out a command line that would let me upload the bootloader?

`K:\Dur2k_D\Download\Programmering\Arduino\ArduinoIDE\hardware\tools\avr/bin/avrdude -CC:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.0/avrdude.conf -v -patmega328p -cusbasp -Pusb -e -Ulock:w:0x3f:m -Uefuse:w:0xfe:m -Uhfuse:w:0xd6:m -Ulfuse:w:0x62:m -B 5
`

As I understand it, the MicroCore is not possible to use with the 328P?

MCUdude, 2018-03-09:
It depends on where you added -B5. When you hit _Burn bootloader_ you're actually issuing two avrdude commands. The first one sets the fuses, and the second one loads the bootloader hex file. You'll need to add -B5 (or -B8, -B16, -B32) to both commands. This can be done by modifying the platform.txt file.

My programmer menu looks like this. I select one of the _slow_ programmers when I'm dealing with clocks < 1 MHz.
[image]

Teddyz, 2018-03-09:
Thanks! My problem is that since the first fails the second command is never shown.
I have now tried to modify almost every platform.txt on my computer. I found yours in: `C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.0\platform.txt`
I guess the line to modify starts with `tools.avrdude.bootloader.pattern=`
But no matter what I change the outcome is the same. I will now try to install MicroCore to see if that gives me the possibility to run USBasp slower. for MiniCore.

Teddyz, 2018-03-09:
Installing MicroCore gave access to more programmers, including slow USBasp, in MiniCore.(sorry for not understanding that from your first instruction).  Now I could flash the bootloader. Without bootloader my program started immediately after boot (of course), now the built in LED flashes twice and there is a 1200ms delay before it starts. Me happy again.

Teddyz, 2018-03-10:
Using MicroCore's USBasp(slow) only works for uploading bootloader and setting fuses. It does not work for uploading sketches. 

USBasp(slow) fails with "avrdude: ser_open(): can't open device "usb": The system cannot find the file specified."
`C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/bin/avrdude -CC:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.0/avrdude.conf -v -patmega328p -carduino -Pusb -B32 -Uhfuse:w:0xd6:m -Ulfuse:w:0x62:m -Uflash:w:D:\Temp\arduino_build_455783/Blink.ino.hex:i`

USBasp fails with "error: program enable: target doesn't answer. 1" (because programmer runs too fast)
`C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/bin/avrdude -CC:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/etc/avrdude.conf -v -patmega328p -cusbasp -Pusb -Uflash:w:D:\Temp\arduino_build_455783/Blink.ino.hex:i `
Could adding '-B32' for any MCU that runs at 1MHz or slower solve this problem? I think these problems and solutions are  worth mentioning in the documentation.

MCUdude, 2018-03-10:
If your USBasp looks like this, you can just short out JP3 in order to physically slow down the clock.
It's generally not a good idea force -B32 when selecting 1 MHz. This may end up with two -B flags (-B2 -B32) for instance, and avrdude would not know what to do.

[image]

Here's the commands that Arduino IDE runs when I select *USBasp (slow)* and burns bootloader for ATmega328P for 1 MHz internal oscillator
```
/Users/Hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino12/bin/avrdude -C/Users/Hans/Documents/Arduino/hardware/MiniCore/avr/avrdude.conf -v -patmega328p -cusbasp -Pusb -B32 -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd6:m -Ulfuse:w:0x62:m 

[Fuses OK, avrdude OK]

/Users/Hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino12/bin/avrdude -C/Users/Hans/Documents/Arduino/hardware/MiniCore/avr/avrdude.conf -v -patmega328p -cusbasp -Pusb -B32 -Uflash:w:/Users/Hans/Documents/Arduino/hardware/MiniCore/avr/bootloaders/optiboot_flash/atmega328p/optiboot_flash_atmega328p_UART0_9600_1000000L.hex:i -Ulock:w:0x0f:m 

[Hex OK, avrdude OK]
```

Heres the command when uploading a program using *USBasp (slow)* for a 328P running at 1 MHz:
```
/Users/Hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino12/bin/avrdude -C/Users/Hans/Documents/Arduino/hardware/MiniCore/avr/avrdude.conf -v -patmega328p -cusbasp -Pusb -B32 -Uhfuse:w:0xd7:m -Ulfuse:w:0x62:m -Uflash:w:/var/folders/2f/bk0yr6zx4290kvkss8ty0gxc0000gn/T/arduino_build_930771/Blink.ino.hex:i 

[Hex loaded, avrdude OK]
```

Teddyz, 2018-03-10:
A jumper in JP3 solved the problem for me. It is very slow but it works reliable.

Flashing bootloader with jumper takes 8 seconds (no matter if I run USBasp or USBasp(slow)

Flashing ArduinoISP (largest example I could find at about 5kB) takes about 72s using USBasp with JP3 attached. Flashing the same code using serial 9600bps took about 17 seconds.

Now I am hopefully finished exploring this (to me) new tool and can start working. :)

Thanks!