MCUdude/MiniCore #42: Difference between reset and power on?

Teddyz, 2018-03-10:
Precondition: A 16 MHz Arduino ProMini-clone with bootloader settings as below and running the example Blink.
[image]

1) When I start by attaching 5V the sketch runs immediately, skipping the bootloader.
2) When I start by releasing the reset button, the led blinks twice and the sketch runs after a second or so.
3) When I start by having the ProMini attached to my CP2102 serial adapter and attach this 'kit' to usb, the bootloader runs exactly seven times and then the sketch starts. 

Number 3 is an odd situation, but I would like to know why bootloader is skipped when attaching power.
Attached a film to show what happens. [film_promini_123.zip](https://github.com/MCUdude/MiniCore/files/1799328/film_promini_123.zip)

MCUdude, 2018-03-10:
1. The bootloader is designed to jump straight into the program code when power is applied. No need to run the bootloader if power was applied and we're waiting for the application to run. The magic happens [here](https://github.com/MCUdude/optiboot_flash/blob/95c5494589d67cea2d48b049f4deb413832db78e/optiboot_flash.c#L493-L541).
2. The bootloader runs one when the reset button is pressed, yes. This is because the USB to serial adapter is connected to the reset pin on the AVR (through a 100n cap), in order for auto reset to work. the dela of 1s is sat [here](https://github.com/MCUdude/optiboot_flash/blob/95c5494589d67cea2d48b049f4deb413832db78e/optiboot_flash.c#L565).
3. This odd (and unnecessary) cycle is caused by the Windows and/or its drivers. It does not happened on my Mac. It's caused by the DTR line on the serial adapter toggling, which cause the auto reset circuitry to trigger. I don't think there's any way we can get rid of this..

Teddyz, 2018-03-10:
Thanks for the explanations. Bypassing the bootloader at power on is a very nice feature that I assumed was a bug since I had never read about it.

closed 2018-03-10 by Teddyz