MCUdude/MiniCore #43: Command Line Build issues

SConaway, 2018-03-13:
The command-line arduino is throwing several errors.

````
$ /Applications/Arduino.app/Contents/MacOS/Arduino --verify --board MiniCore:avr:168:variant=modelP --verbose --preserve-temp-files src/src.ino
Already up to date.
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/steven/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/steven/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/steven/Documents/Arduino/libraries -fqbn=MiniCore:avr:168:variant=modelP -ide-version=10805 -build-path /var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_544745 -warnings=none -build-cache /var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_cache_614830 -prefs=build.warn_data_percentage=75 -verbose /Users/steven/Outdoor-Lighting-Controller/Arduino_Nano/src/src.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/steven/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/steven/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/steven/Documents/Arduino/libraries -fqbn=MiniCore:avr:168:variant=modelP -ide-version=10805 -build-path /var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_544745 -warnings=none -build-cache /var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_cache_614830 -prefs=build.warn_data_percentage=75 -verbose /Users/steven/Outdoor-Lighting-Controller/Arduino_Nano/src/src.ino
Using board '168' from platform in folder: /Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0
Using core 'MCUdude_corefiles' from platform in folder: /Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0
Detecting libraries used...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega168p -DF_CPU= -DARDUINO=10805 -DARDUINO_AVR_ATmega168 -DARDUINO_ARCH_AVR   "-I/Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0/cores/MCUdude_corefiles" "-I/Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0/variants/standard" "/var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_544745/sketch/src.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega168p -DF_CPU= -DARDUINO=10805 -DARDUINO_AVR_ATmega168 -DARDUINO_ARCH_AVR   "-I/Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0/cores/MCUdude_corefiles" "-I/Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0/variants/standard" "/var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_544745/sketch/src.ino.cpp" -o "/var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_544745/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_544745/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega168p -DF_CPU={build.f_cpu} -DARDUINO=10805 -DARDUINO_AVR_ATmega168 -DARDUINO_ARCH_AVR   "-I/Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0/cores/MCUdude_corefiles" "-I/Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0/variants/standard" "/var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_544745/sketch/src.ino.cpp" -o "/var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_544745/sketch/src.ino.cpp.o"
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/util/delay.h: In function 'void _delay_ms(double)':
<command-line>:0:8: error: 'build' was not declared in this scope
In file included from /Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0/cores/MCUdude_corefiles/USBAPI.h:27:0,
                 from /Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0/cores/MCUdude_corefiles/Arduino.h:289,
                 from /var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_544745/sketch/src.ino.cpp:1:
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/util/delay.h:174:21: error: invalid operands of types 'void' and 'double' to binary 'operator/'
  __tmp = ((F_CPU) / 1e3) * __ms;
                     ^
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/util/delay.h: In function 'void _delay_us(double)':
<command-line>:0:8: error: 'build' was not declared in this scope
In file included from /Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0/cores/MCUdude_corefiles/USBAPI.h:27:0,
                 from /Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0/cores/MCUdude_corefiles/Arduino.h:289,
                 from /var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_544745/sketch/src.ino.cpp:1:
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/util/delay.h:263:21: error: invalid operands of types 'void' and 'double' to binary 'operator/'
  __tmp = ((F_CPU) / 1e6) * __us;
                     ^
exit status 1

$ /Applications/Arduino.app/Contents/MacOS/Arduino --verify --board MiniCore:avr:328 --verbose --preserve-temp-files src/src.ino
Already up to date.
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/steven/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/steven/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/steven/Documents/Arduino/libraries -fqbn=MiniCore:avr:168 -ide-version=10805 -build-path /var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_155433 -warnings=none -build-cache /var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_cache_842587 -prefs=build.warn_data_percentage=75 -verbose /Users/steven/Outdoor-Lighting-Controller/Arduino_Nano/src/src.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/steven/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/steven/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/steven/Documents/Arduino/libraries -fqbn=MiniCore:avr:168 -ide-version=10805 -build-path /var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_155433 -warnings=none -build-cache /var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_cache_842587 -prefs=build.warn_data_percentage=75 -verbose /Users/steven/Outdoor-Lighting-Controller/Arduino_Nano/src/src.ino
Using board '168' from platform in folder: /Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0
Using core 'MCUdude_corefiles' from platform in folder: /Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0
Detecting libraries used...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu= -DF_CPU= -DARDUINO=10805 -DARDUINO_AVR_ATmega168 -DARDUINO_ARCH_AVR   "-I/Use [... cut ...]
In file included from /Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0/cores/MCUdude_corefiles/USBAPI.h:27:0,
                 from /Users/steven/Library/Arduino15/packages/MiniCore/hardware/avr/2.0.0/cores/MCUdude_corefiles/Arduino.h:289,
                 from /var/folders/ry/9vqr33vj43n6yv55nzr7plkr0000gn/T/arduino_build_433022/sketch/src.ino.cpp:1:
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/util/delay.h:263:21: error: invalid operands of types 'void' and 'double' to binary 'operator/'
  __tmp = ((F_CPU) / 1e6) * __us;
                     ^
exit status 1
````

per1234, 2018-03-13:
You didn't specify the full fqbn (fully qualified board name) in your command. In this case the most problematic missing element being `clock`, but I'd just do the full thing to be safe.

The easiest way to get the correct fqbn is:

1. File > Preferences > Show verbose output during: > compilation (check) > OK
2. Select the exact board configuration you want via the various Tools submenus
3. Sketch > Verify/compile
4. After the compilation finishes, scroll the black console window at the bottom of the Arduino IDE window all the way to the top. Somewhere along the first line you will see something like this:

```
-fqbn=MiniCore:avr:168:bootloader=true,variant=modelP,BOD=2v7,LTO=Os,clock=16MHz_external
```
So in this case the fqbn you should pass to arduino via the `--board` option is:
```
MiniCore:avr:168:bootloader=true,variant=modelP,BOD=2v7,LTO=Os,clock=16MHz_external
```
Thus the updated command would be:
/Applications/Arduino.app/Contents/MacOS/Arduino --verify --board MiniCore:avr:168:bootloader=true,variant=modelP,BOD=2v7,LTO=Os,clock=16MHz_external --verbose --preserve-temp-files src/src.ino
```

SConaway, 2018-03-13:
Thank you. Wasn't sure how to get the fqbn.

closed 2018-03-13 by SConaway