MCUdude/MiniCore #44: #error "No SLEEP mode defined for this device."

opensesame11, 2018-03-16:
I am compining for ATmega328PB using MiniCore 2.0.0, Arduino 1.8.6, and Arduino toolchain 1.6.21. The problem I am experiencing also happened on my older copy of Arduino 1.8.5 before I updated, with an unknown version of the toolchain.

When I compile any program, even just a simple "void setup(){} void loop(){}", it complains about this. The observed error of "#error "No SLEEP mode defined for this device."" only occurs with the build target set to the PB variation. Both the 328(A) and 328P(A) targets compile fine. 

The header files that are linked to the error are as follows:
hardware\tools\avr\avr\include\avr\sleep.h:224:6,
which is called by MiniCore\hardware\avr\2.0.0\cores\MCUdude_corefiles/wiring_extras.h:14:0,
which in turn is called by MiniCore\hardware\avr\2.0.0\cores\MCUdude_corefiles/Arduino.h:290

I imagine there is probably a typo in one of the #DEFINE lines somewhere in the core, but I don't myself know where to look.

opensesame11, 2018-03-16:
From looking in the sleep.h header file, I can see where this compiler error is being thrown. For sleep.h to be preprocessed properly, the core has to define the sleep mode as SM, SM1, or SM2. I tried adding #define SM2 to my .ino file, but it looks like the inclusion of these headers occurs, surprise surprise, before the user source. I'll take a crack at putting that definition somewhere in the MiniCore files to see if that fixes anything.

per1234, 2018-03-16:
There's an error in MiniCore's instructions regarding the PB chips. You need to do this:

1. File > Preferences > Additional Boards Manager URLs > https://downloads.arduino.cc/packages/package_avr_3.6.0_index.json
2. Tools > Board > Boards Manager
3. Wait for downloads to finish
4. Click "Arduino AVR Boards"
5. From the dropdown version menu, select "1.6.206"
6. Click "Install"
7. Wait for the installation to finish
8. Click "Close"

After that the problem should be solved. I'll submit a pull request to update the readme.

closed 2018-03-16 by MCUdude