MCUdude/MiniCore #48: Firmware is much bigger with MiniCore comparing to std Arduino 1.8.5

latonita, 2018-03-28:
Hello.
I understand that original goals of the project were a bit different, however I couldn't notice, that in case I use MiniCore I'm getting much less free memory.

Here is two of my projects I just tried to compile. Is this normal? Please advise.
Thank you.

### https://github.com/latonita/arduino-canbus-monitor
| Board | Program size | Diff | Variables | Diff |
|-------|--------------|------|-----------|------|
| Arduino Pro Mini 168 | 9784 bytes (68%) |  . | 526 bytes (51%) |  . |
| MicroCore 168 | 11716 bytes (73%) | :-1: -1932 bytes | 538 bytes (52%) | :-1: -12 bytes |

### https://github.com/latonita/jeep-canbus  ->  /jeep-aux-and-extra-features/
| Board | Program size | Diff | Variables | Diff |
|--------|----------------|-----|-----------|------|
| Arduino Pro Mini 168 | 6072 bytes (42%) |  .  | 634 bytes (61%) |  . |
| MicroCore 168 | 8268 bytes (52%) | :-1: -2196 bytes | 646 bytes (63%) | :-1: -12 bytes |

per1234, 2018-03-28:
Try this:
Tools > Compiler LTO > Enabled

Arduino AVR Boards uses LTO but MiniCore has it disabled by default to provide compatibility with older compiler versions that don't support LTO.

latonita, 2018-03-28:
Thanks, its much better now, but still more than in "standard". Need to compare build options.

### https://github.com/latonita/arduino-canbus-monitor
| Board | Program size | Diff | Variables | Diff |
|-------|--------------|------|-----------|------|
| Arduino Pro Mini 168 | 9784 bytes (68%) |  . | 526 bytes (51%) |  . |
| MicroCore 168 | 11716 bytes (73%) | :-1: -1932 bytes | 538 bytes (52%) | :-1: -12 bytes |
| MicroCore 168 LTO | 10108 bytes (63%) | :-1: -324 bytes | 526 bytes (51%) | :ok_hand: |

### https://github.com/latonita/jeep-canbus  ->  /jeep-aux-and-extra-features/
| Board | Program size | Diff | Variables | Diff |
|--------|----------------|-----|-----------|------|
| Arduino Pro Mini 168 | 6072 bytes (42%) |  .  | 634 bytes (61%) |  . |
| MicroCore 168 | 8268 bytes (52%) | :-1: -2196 bytes | 646 bytes (63%) | :-1: -12 bytes |
| MicroCore 168 LTO | 6396 bytes (40%) | :-1: -324 bytes | 634 bytes (61%)| :ok_hand: |

Tried to compare build options just analysing output. Several stages. I have left only different options:
### Detecting libraries used...
STD:avr-g++ -fpermissive
MIN:avr-g++ -Wextra 

### Compiling sketch...
STD: avr-g++ -fpermissive -flto 
MIN: avr-g++ 

### Linking everything together...
STD:avr-gcc -g -fuse-linker-plugin 
MIN:avr-gcc -w

per1234, 2018-03-28:
I got the same numbers as you for arduino-canbus-monitor (didn't try the other sketch), then replaced MiniCore's SoftwareSerial library with the [Arduino AVR Core's SoftwareSerial library](https://github.com/arduino/ArduinoCore-avr/tree/master/libraries/SoftwareSerial) and got 9752/522 bytes.

Note that when you change the library, the Arduino IDE wants to still use the cached version of the library (I think because the timestamps on the Arduino AVR Boards library are older). You can clear the cache by changing boards for a compilation or restarting the IDE.

MCUdude, 2018-03-29:
I'm om vaccation right now, so I'm not able to dig into this right now. @per1234 how can the MiniCore SoftwareSerial library cause the compiled code to be 324 bytes larger that the stock library?

per1234, 2018-03-30:
The reason for the larger size is that MiniCore has this else clause in SoftwareSerial::begin, which the Arduino AVR Boards SoftwareSerial library doesn' have:
https://github.com/MCUdude/MiniCore/blob/master/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp#L378-L420

What I can't figure out is why that else clause is being executed for the arduino-canbus-monitor sketch since it [sets the SoftwareSerial RX pin to 8](https://github.com/latonita/arduino-canbus-monitor/blob/master/can-232.cpp#L26), which is a PCINT on the ATmega168 so the if statement should be true and that else clause should not be entered.

MCUdude, 2018-03-30:
> The reason for the larger size is that MiniCore has this else clause in SoftwareSerial::begin, which the Arduino AVR Boards SoftwareSerial library doesn't have

Using "regular" interrupts only makes sense for microcontrollers without PCINT on all pins, Which is ATmega8, ATmega16, ATmega32, ATmega64, ATmega128, ATmega640, ATmega1280, ATmega1281, ATmega2560 and ATmega2561. I should probably get around and insert some #ifdefs

commit 8bb566f3 (2018-04-01): Improve SoftwareSerial library
This fix should improve the compiled code size for microcontrollers with and without PCINTs. Closes #48
closed 2018-04-01 by MCUdude