MCUdude/MiniCore #49: Totally not an issue

LMDIGroup, 2018-03-30:
Was trying to find a way to contact you directly but...could not.

I am wanting to modify the pins_arudino.h file to make the pin outs correspond with the physical pin numbers of the chip. (explanation: I was teaching a class with chips on board and the most common issue was students using the wrong pin numbers in their code. Since, I have very limited time in my class I thought "Why not just make the pin numbers correspond to the physical pin numbers since it is all abstracted anyway?"

I took a look and tried to modify your pins_arduino.h file but could not get a working version.

I tired to modify the terniery equations of analogInputToDigitalPin(p) & analogPinToChannel(p) and digitalPinHasPWM(p) as well as the digital #defines. I also resorted the digital_pin_to_port_PGM and digital_pin_to_bit_mask_PGM but I was very uncertain if those were the correct things to be modifying.

if you could please point me in the right direction I would sincerely appreciate it. You can contact me directly if you wish teach4th (at) lmdigroup (dot com)

MCUdude, 2018-03-31:
Do you want the pins in to be mapped to the physical pins of the 28 pin DIP package, or do you want to use the port registers, such as PB0 and PB1?

I think it's easier if you leave the pins_arduino.h file and add your new pinout using macros.

LMDIGroup, 2018-03-31:
We are using the 28 pin DIP ATmega328P. I originally wanted to name the pins after the physical pin numbers of the chip 1-28 counter clockwise starting with Reset on pin 1. I understand about the macro thing I could have them use some name and just #include a header that #defines the pins to the arduino pin numbers/names but then when referencing the help documentation it would not make sense to them.

Anyway I appreciate the response and perhaps I can keep fiddling with and googling the pins_arduino.h file until it happens to work.

Thanks.

MCUdude, 2018-04-01:
It's totally possible to create a pins_arduino.h file But you'll "waste" some additional space, due to the fact that the pin number you're using is the corresponding place in the digital_pin_to_port_PGM,  digital_pin_to_bit_mask_PGM and digital_pin_to_timer_PGM array.

The `digital_pin_to_port_PGM` maps the physical port to to each Arduino pin. Arduino pin 0 - 7 uses PORTD, that's why PD appears on the first eight lines (0 to 7).

The `digital_pin_to_bit_mask_PGM` maps the port bit numbers to each Arduino pin. The _BV macro is defined [here](https://www.microchip.com/webdoc/AVRLibcReferenceManual/group__avr__sfr_1ga11643f271076024c395a93800b3d9546.html).
 If you look at the [ATmega328p pinout](https://github.com/MCUdude/MiniCore#pinout) you can see that Arduino pin 0 - 7 is mapped to PD0 -PD7 (PORTD 0 - PORTD 7). That's why the first line (place zero) has _BV(0), and the eight line has _BV(7).

The `digital_pin_to_timer_PGM` is a little more tricky since there's a macro for ATmega8 in there. You can safely delete what's inside the #ifdef, and keep what's in the #else, since you're dealing with ATmega328P.
You can read the ATmega328P datasheet or use the pinout pic in order to figure out which pins uses which timer.

What may be a little tricky is that some pins (physical pin 7 and 8) isn't used, but will always have a place in the three arrays. That's why you'll need to replace The content of these lines with a placeholder that wan't screw up anything. You can use the macros `NOT_A_PORT`, `NOT_A_PIN`, and `NOT_ON_TIMER` for the lines that shouldn't be used.

closed 2018-04-28 by MCUdude