MCUdude/MiniCore #50: Native EEPROM library is not working with ATmega328

AyrtonEstrella, 2018-04-02:
Native EEPROM library is not working when I select the ATmega328 with any of its variants.

#include <EEPROM.h>
                    ^
compilation terminated.
exit status 1
Error compiling for board ATmega328.

[image]

per1234, 2018-04-02:
Duplicate of https://github.com/MCUdude/MiniCore/issues/47

This issue has already been resolved in https://github.com/MCUdude/MiniCore/commit/a920f2c7654c992c3c163ad0870425b4352f5d6f but there has not been a release since that time. If you like, you can do a manual installation of the beta version of MiniCore, which will resolve this issue:
https://github.com/MCUdude/MiniCore#manual-installation

MCUdude, 2018-04-28:
EEPROM should now work with the latest boards manager version (v2.0.1)

closed 2018-04-28 by MCUdude