MCUdude/MiniCore #52: Cannot compile bootloader

canique, 2018-05-01:
The documentation says:
"You need to install make, avr-libc, avr-binutils and avr-gcc"

I have
avr-libc (2.0.0)
binutils-avr (2.26)
gcc-avr (5.4.0)

When executing:
`/usr/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega328pb -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=2    -DLED=B5  -DUART=0   -c -o optiboot_flash.o `optiboot_flash.c``

I get numerous errors...
`In file included from optiboot_flash.c:250:0:
/usr/lib/avr/include/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
 #    warning "device type not defined"
      ^
In file included from optiboot_flash.c:252:0:
/usr/lib/avr/include/avr/eeprom.h:41:3: warning: #warning "Device does not have EEPROM available." [-Wcpp]
 # warning "Device does not have EEPROM available."
   ^
In file included from optiboot_flash.c:259:0:
boot.h:127:4: error: #error AVR processor does not provide bootloader support!
 #  error AVR processor does not provide bootloader support!
    ^
boot.h:138:4: error: #error Cannot find SPM Enable bit definition!
 #  error Cannot find SPM Enable bit definition!
    ^
optiboot_flash.c:311:2: warning: #warning BAUD_RATE error greater than 2% [-Wcpp]
 #warning BAUD_RATE error greater than 2%
  ^
optiboot_flash.c: In function ‘main’:
optiboot_flash.c:506:8: error: ‘MCUSR’ undeclared (first use in this function)
   ch = MCUSR;
        ^
optiboot_flash.c:506:8: note: each undeclared identifier is reported only once for each function it appears in
In file included from /usr/lib/avr/include/avr/io.h:99:0,
                 from optiboot_flash.c:250:
optiboot_flash.c:521:20: error: ‘WDRF’ undeclared (first use in this function)
     if ((ch & (_BV(WDRF) | _BV(EXTRF))) != _BV(EXTRF)) { 
                    ^
optiboot_flash.c:521:32: error: ‘EXTRF’ undeclared (first use in this function)
     if ((ch & (_BV(WDRF) | _BV(EXTRF))) != _BV(EXTRF)) { 
                                ^
optiboot_flash.c:545:3: error: ‘TCCR1B’ undeclared (first use in this function)
   TCCR1B = _BV(CS12) | _BV(CS10); // div 1024
   ^
In file included from /usr/lib/avr/include/avr/io.h:99:0,
                 from optiboot_flash.c:250:
optiboot_flash.c:545:16: error: ‘CS12’ undeclared (first use in this function)
   TCCR1B = _BV(CS12) | _BV(CS10); // div 1024
                ^
optiboot_flash.c:545:28: error: ‘CS10’ undeclared (first use in this function)
   TCCR1B = _BV(CS12) | _BV(CS10); // div 1024
                            ^
In file included from optiboot_flash.c:269:0:
pin_defs.h:231:18: error: ‘UCSR0A’ undeclared (first use in this function)
 #define UART_SRA UCSR0A
                  ^
optiboot_flash.c:557:3: note: in expansion of macro ‘UART_SRA’
   UART_SRA = _BV(U2X0); //Double speed mode USART0
   ^
In file included from /usr/lib/avr/include/avr/io.h:99:0,
                 from optiboot_flash.c:250:
optiboot_flash.c:557:18: error: ‘U2X0’ undeclared (first use in this function)
   UART_SRA = _BV(U2X0); //Double speed mode USART0
                  ^
In file included from optiboot_flash.c:269:0:
pin_defs.h:232:18: error: ‘UCSR0B’ undeclared (first use in this function)
 #define UART_SRB UCSR0B
                  ^
[... 174 lines cut ...]
boot.h:207:60: error: ‘PGERS’ undeclared (first use in this function)
 #define __BOOT_PAGE_ERASE         (_BV(__SPM_ENABLE) | _BV(PGERS))
                                                            ^
optiboot_flash.c:983:38: note: in expansion of macro ‘__BOOT_PAGE_ERASE’
      do_spm((uint16_t)(void*)address,__BOOT_PAGE_ERASE,0);
                                      ^
boot.h:208:60: error: ‘PGWRT’ undeclared (first use in this function)
 #define __BOOT_PAGE_WRITE         (_BV(__SPM_ENABLE) | _BV(PGWRT))
                                                            ^
optiboot_flash.c:999:38: note: in expansion of macro ‘__BOOT_PAGE_WRITE’
      do_spm((uint16_t)(void*)address,__BOOT_PAGE_WRITE,0);
                                      ^
optiboot_flash.c: In function ‘do_spm’:
optiboot_flash.c:1084:25: error: ‘__SPM_REG’ undeclared (first use in this function)
     : "i" (_SFR_IO_ADDR(__SPM_REG)),
                         ^
boot.h:199:65: error: ‘__SPM_ENABLE’ undeclared (first use in this function)
 #define boot_spm_busy()               (__SPM_REG & (uint8_t)_BV(__SPM_ENABLE))
                                                                 ^
boot.h:205:49: note: in expansion of macro ‘boot_spm_busy’
 #define boot_spm_busy_wait()          do{}while(boot_spm_busy())
                                                 ^
optiboot_flash.c:1095:5: note: in expansion of macro ‘boot_spm_busy_wait’
     boot_spm_busy_wait();
     ^

MCUdude, 2018-05-01:
That's because the stock version of avr-gcc 5.4.0 doesn't support ATmega328PB. I'm using the compiler that's included in AVR boards v1.6.207. On my mac it's located in the following directory:
`/Users/Hans/Library/Arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/avr-gcc`

canique, 2018-05-02:
Great, thank you!
I could compile it under Windows now by using the compiler in /Users/{username}/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/avr-gcc

Just for reference:
Under linux I used this preferences file under /etc/apt/preferences.d/:
`Package: gcc-avr binutils-avr avr-libc libmpfr6 libstdc++6 libmpc3
Pin: release a=testing
Pin-Priority: 950

Package: *
Pin: release a=testing
Pin-Priority: 99`

and installed with this command
`sudo apt-get install gcc-avr binutils-avr avr-libc libmpfr6 libmpc3`

but this only gives you version 5.4.0-atmel3.6.0 so it's not the correct version.

MCUdude, 2018-05-02:
Thanks for the info. I'll update the instructions as soon as gcc 5.4.x supports the new *PB series 👍

MCUdude, 2019-03-18:
Should be fixed now. See [https://github.com/MCUdude/optiboot_flash](https://github.com/MCUdude/optiboot_flash) for details.

closed 2019-03-18 by MCUdude