MCUdude/MiniCore #53: Error with sleep when using 328pb variant only

yaddanadda, 2018-05-23:
[code]Arduino: 1.8.4 (Windows 10), Board: "ATmega328, No, 328PB, 2.7v, Disabled (default), 8 MHz internal"

Warning: Board breadboard:avr:atmega328bb doesn't define a 'build.board' preference. Auto-set to: AVR_ATMEGA328BB
In file included from C:\Users\Lion.O\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.1\cores\MCUdude_corefiles/wiring_extras.h:14:0,

                 from C:\Users\Lion.O\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.1\cores\MCUdude_corefiles/Arduino.h:290,

                 from sketch\Genesis_Intellivision.ino.cpp:1:

c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\sleep.h:224:6: error: #error "No SLEEP mode defined for this device."

     #error "No SLEEP mode defined for this device."

      ^

exit status 1
Error compiling for board ATmega328.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.[/code]

I am using this minicore since Arduino IDE does not directly support the newer ATMega168pb and 328pb (they were released just a couple years ago).  2 extra IO pins, better handling of A6 and A7 pins (they were multiplexed on older ATMega x8's and can only be used as analog input), and the minicore also lets me set internal clock which frees up XTAL pins for 2 more digital IO pins.

I am not using sleep anywhere in my code and when I change the variant to 328 or 328p the code compiles fine but when I use 328pb I get error about sleep.  Oddly enough if I used variant 168, 168p, and 168pb (and all 3 88's as well, program too large for 48's and 8's) they compile fine as well.  Why won't 328pb work?  I tried 328pb with external and internal clock, both the same result.  It's as if there's a bad line in 328pb that handles sleep?

per1234, 2018-05-23:
You forgot to follow these instructions:
https://github.com/MCUdude/MiniCore#atmega4888168328pb-series

yaddanadda, 2018-05-23:
Hmmm I thought it was installed but I probably forgot to copy that link when I copied minicore link.  Added it and installed, now the error goes away.

yaddanadda, 2018-05-23:
closing this, error was ID-10-T

closed 2018-05-23 by yaddanadda