MCUdude/MiniCore #54: Blink_using_Timer0 does not work with Atmega8: OCR0 not available

shaffenmeister, 2018-05-28:
**Observation:**
Compilation of Blink_using_Timer0.ino fails for Atmega8.

**Error message:**
```
/home/cappel/.arduino15/packages/MiniCore/hardware/avr/2.0.1/libraries/AVR_examples/examples/Blink_using_Timer0/Blink_using_Timer0.ino: In function 'int main()':
Blink_using_Timer0:29: error: 'OCR0' was not declared in this scope
     OCR0 = 0x00;  // Output compare
     ^
exit status 1
'OCR0' was not declared in this scope
```
**Possible reason:**
OCR0 ist not available in Atmega8(L), according to Atmel data sheet.

**Software environment:**
Arduino 1.8.5
MiniCore 2.0.1
BSP: Arduino AVR core 1.6.207
Xubuntu 18.04, Windows 10

mieperDE, 2018-06-15:
can confirm
I checked the data sheet. It looks like the Atmega8 doesn't have that register.

MCUdude, 2018-06-17:
Thanks for the reminder, I totally forgot to look at it. It correct that ATmega8 doesn't have that register. Will fix it soon 🙂

commit e1a8ca20 (2018-06-17): ATmega8 doesn't have an OCR0 register
Closes #54
closed 2018-06-17 by MCUdude