MCUdude/MiniCore #57: i2c blocks indefinitely if SDA line is shorted to ground on ATmega328

macdonaldtomw, 2018-06-21:
I'm using the Wire library on my ATmega328 (board version 328) as a master.  Configuration is shown in following picture:

[image]

I accidentally had the SDA line shorted to ground.  The ensuing behaviour of the ATmega328 seems strange, and I think it may be a bug.  Not sure if its a bug from the MiniCore project or from the Arduino Wire library in general.

Basically what happens in this case is that the ATmega328 just keeps sending clock pulses out indefinitely.

The expected behaviour should be that the ATmega328 sees the SDA line is low when it goes to use the i2c bus as a master, and concludes that another master has control of the line and returns immediately with an `error code 4 - other error`

It seems that there is no "bus is busy" error described in the Arduino `Wire.endTransmission()` return codes defined here https://www.arduino.cc/en/Reference/WireEndTransmission

This makes me think that the Arduino Wire library is not compatible with the i2c hardware spec, which says that any master attempting to take control of the bus should check to make sure that the bus is not already busy, as described here:  https://www.i2c-bus.org/multimaster/

MCUdude, 2018-06-21:
Thanks for reporting. Can you reproduce this issue if you choose **Arduino UNO** as your board?

MCUdude, 2018-07-14:
The Wire library is simply forked of the official Arduino repo. If you believe it doesn't follow the i2c standard then you should open an issue there instead.

closed 2018-07-14 by MCUdude