MCUdude/MiniCore #59: 7.372800 and 14.745600 MHz clock

berkesandor, 2018-06-24:
Will there be available bootloaders for quartzes of 7.372800 and 14.745600 MHz?
This would be a great help to reduce communication errors.
Thank you for your answer.
(sorry for my bad English knowledge)

MCUdude, 2018-06-24:
Hi!
I don't think I'll add these clock frequencies to the Clock menu in Arduino IDE, I'm working on a new build script for Optiboot_flash so that I can easily add new clock frequencies and baud rates. I'll push the new compiled bootloader files soon 👍

MCUdude, 2018-06-24:
You'll find everything you need here: [https://github.com/MCUdude/optiboot_flash](https://github.com/MCUdude/optiboot_flash)

closed 2018-06-24 by MCUdude

berkesandor, 2018-06-24:
How can I use this hex file in the Arduino IDE?

MCUdude, 2018-06-25:
The changes are now pushed to this repo.
* Do a manual install of MiniCore
* Open the boards.txt file
* Add a new [clock frequency entry](https://github.com/MCUdude/MiniCore/blob/master/avr/boards.txt#L103-L107)
* Paste the lines below
* Restart Arduino IDE

```
328.menu.clock.7_3728MHz_external=7.3728 MHz external
328.menu.clock.7_3728MHz_external.upload.port=UART0
328.menu.clock.7_3728MHz_external.upload.speed=115200
328.menu.clock.7_3728MHz_external.bootloader.low_fuses=0b1111{bootloader.cksel_bits}
328.menu.clock.7_3728MHz_external.build.f_cpu=7372800L

328.menu.clock.14_7456MHz_external=14.7456 MHz external
328.menu.clock.14_7456MHz_external.upload.port=UART0
328.menu.clock.14_7456MHz_external.upload.speed=115200
328.menu.clock.14_7456MHz_external.bootloader.low_fuses=0b1111{bootloader.cksel_bits}
328.menu.clock.14_7456MHz_external.build.f_cpu=14745600L
```