MCUdude/MiniCore #60: Can't upload to ATmega328PB

macdonaldtomw, 2018-07-16:
I'm using the following setup:

Arduino IDE V1.8.5
Arduino AVR Boards (board manager) v1.6.207
MiniCore v2.0.1

Using [AVR pocket programmer](https://www.sparkfun.com/products/9825) from SparkFun

to program a ATmega328PB mounted on a ATmega328PB Xplained development board

Compiling code in Arduino IDE works fine.

When I go to "burn bootloader", I get the following message:

```
Arduino: 1.8.5 (Windows 10), Board: "ATmega328, No, 328PB, 2.7v, Disabled (default), 8 MHz internal"

C:\Users\Boompy\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/bin/avrdude -CC:\Users\Boompy\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.1/avrdude.conf -v -patmega328pb -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd7:m -Ulfuse:w:0xe2:m 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Boompy\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.1/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x000000 (retrying)

Error while burning bootloader.
Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

If I open a terminal and type `avrdude -c usbtiny -p m328p`

I get the following:

```
C:\Users\Boompy>avrdude -c usbtiny -p m328p

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9516
avrdude: Expected signature for ATMEGA328P is 1E 95 0F
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.
```

What am I doing wrong?  How can I use this setup to successfully work with the ATmega328PB?

macdonaldtomw, 2018-07-16:
Update:

I just unpackaged a fresh ATmega328PB Xplained board, and used the same process as above.  Totally different result.  Here is what I get when I burn bootloader:

```
C:\Users\Boompy\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/bin/avrdude -CC:\Users\Boompy\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.1/avrdude.conf -v -patmega328pb -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd7:m -Ulfuse:w:0xe2:m 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Boompy\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.1/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9516 (probably m328pb)
avrdude: erasing chip
avrdude: Using SCK period of 10 usec
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

C:\Users\Boompy\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/bin/avrdude -CC:\Users\Boompy\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.1/avrdude.conf -v -patmega328pb -cusbtiny -Uflash:w:C:\Users\Boompy\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.1/bootloaders/empty/empty.hex:i -Ulock:w:0x0f:m 
Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s
[... 119 lines cut ...]

avrdude: verifying ...
avrdude: 2 bytes of flash verified
avrdude: reading input file "0x0f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0f:
avrdude: load data lock data from input file 0x0f:
avrdude: input file 0x0f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.
```

so.... the only differences I can think of here is that on the first Xplained board that I tried, I was using Atmel Studio for a while and using the integrated mEDBG chip on that board to program/debug the ATmega328PB before switching over to Arduino IDE to try and continue development there.

Any idea why using Atmel Studio renders the ATmega328PB un-programmable by the Arduino IDE?

MCUdude, 2018-07-16:
It seems like the SPI lines are blocked or you have a bad connection.

I have a 328PB board myself (not the Xplained board), and I've never experienced any issues regarding burning bootloader or uploading.

macdonaldtomw, 2018-07-16:
I finally figured out what the issue is.

Apparently, once you have flashed a build configured to use mEDGB on-chip debugger, any subsequent builds made for ISP uploading will fail miserably, and avrdude won't even be able to read the correct device signature.

In order to fix the problem, you need to use Atmel Studio to upload a test app using the _Start Debugging and Break_ option, then go to _Tools -> Select Profile_ and select _Advanced_ option.  This will change the GUI such that a new option appears in the _Debug_ menu called __Disable debugWire and close_.  

Afterwards, you should be able to select the _ISP_ option in the _Programmer Tools/Config_ page.

See here for more details:

https://www.avrfreaks.net/forum/cant-get-out-debug-mode#comment-2512496

MCUdude, 2018-07-17:
Great, thanks for providing a solution for other users who may experience the same problem!

closed 2018-07-17 by MCUdude