MCUdude/MiniCore #62: Problem importing MiniCore Arduino project into Atmel Studio 7

macdonaldtomw, 2018-07-17:
Atmel Studio 7 supports directly importing Arduino sketches.  It automatically creates the makefiles, and imports the core files associated with the board you are using.

Unfortunately, it appears that it fails to import for that MiniCore targets.  For example, here is what happens when I try to import an Arduino sketch targeted at the ATmega328 (see below video).

[Trying to import ATmega328PB MiniCore Arduino project into Atmel Studio](https://youtu.be/GzcRKkcIa-c)

As you can see, in the import dialogue titled _Creat C++ project from Arduino Sketch_ , after selecting the _ATmega328_ MiniCore board option, the _Device_ dropdown lists a single cryptic entry: `Arduino.IDE.Board.Service.McuVariant`

I think this is probably a big clue as to why this is failing.

Is there a workaround for this?  I have already tried copying/replacing files from MiniCore master to the Atmel Studio project, however this doesn't seem to work well with the toolchain settings, and won't compile.  See video here for that attempt:

[Trying to replace Arduino core files after importing wrong device type to Atmel Studio](https://youtu.be/fVSeHf0HmuE)

MCUdude, 2018-07-17:
Sorry, but none of your videos works. They're all back boxes, not even a play button.

Anyways I'm pretty sure Atmel studio don't support MiniCore. Personally I'm using [Visual Micro](http://www.visualmicro.com) when using MiniCore with Atmel Studio.

macdonaldtomw, 2018-07-18:
Visual Micro seems to have done the trick.  Thanks!

closed 2018-07-18 by macdonaldtomw