MCUdude/MiniCore #63: digitalPinToInterrupt(21) returns 255 with ATmega328PB

macdonaldtomw, 2018-07-18:
I'm using Visual Micro along with Atmel Studio, using mEDBG debugger on an Xplained Mini 328PB (using internal 8MHz oscillator)

I'm trying to get the following sketch to work:

```
const byte ledPin = 13;
const byte interruptPin = 21;
volatile byte state = LOW;

void setup() {
	pinMode(ledPin, OUTPUT);
	pinMode(interruptPin, INPUT_PULLUP);
	attachInterrupt(digitalPinToInterrupt(interruptPin), blink, CHANGE);
}

void loop() {
	digitalWrite(ledPin, state);
}

void blink() {
	state = !state;
}
```

If I build a debugging version of this sketch, and step through it, it appears that the call to `digitalPinToInterrupt(21)` returns 255, which makes the `attachInterrupt()` call silently fail.

See video for step-debugging:

https://youtu.be/QOlM6OYCwww

I think ultimately this is a problem with the `digitalPinToInterrupt()` function in `pins_arduino.h`  which uses the following implementation of `digitalPinToInterrupt()`:

```
#elif defined(__AVR_ATmega328PB__)
#define digitalPinToPCICR(p)     (((p) >= 0 && (p) <= 26) ? (&PCICR) : ((uint8_t *)0))
#define digitalPinToPCICRbit(p)  (((p) <= 7) ? 2 : (((p) <= 13 || (p) == 20 || (p) == 21) ? 0 : (((p) <= 22) ? 1 : 2)))
#define digitalPinToPCMSK(p)     (((p) <= 7) ? (&PCMSK2) : (((p) <= 13 || (p) == 20 || (p) == 21) ? (&PCMSK0) : (((p) <= 22) ? (&PCMSK1) : (((p) <= 26) ? (&PCMSK3) : ((uint8_t *)0)))))
#define digitalPinToPCMSKbit(p)  (((p) <= 7) ? (p) : (((p) <= 13) ? ((p) - 8) : ((p) <= 21) ? ((p) - 14) : ((p) == 22) ? ((p) - 16) : ((p) - 23)))
#define digitalPinToInterrupt(p) ((p) == 2 ? 0 : ((p) == 3 ? 1 : NOT_AN_INTERRUPT))
#endif
```

MCUdude, 2018-07-18:
The arduino API doesn't really support PCINTs out of the box. That means digitalPinToInterrupt() will only work if you insert the number 2 or 3 in it. 

`NOT_AN_INTERRUPT` [is defined as -1](https://github.com/MCUdude/MiniCore/blob/ed106a31f6565ab4000b93129eabfea37929d99f/avr/cores/MCUdude_corefiles/Arduino.h#L216), which will become 255 if you store the value to an unsigned variable.

Try a google search for *Arduino PCINT library*

macdonaldtomw, 2018-07-18:
Okay, thanks.  I'm just rolling up my sleeves and doing it in Atmel Studio with avrlibc.

Thanks again!

closed 2018-07-19 by MCUdude