MCUdude/MiniCore #64: Caveat about micros() does not match reality

SpenceKonde, 2018-07-18:
In the documentation, users are warned about clock speeds by which 64 is not evenly divisible, and told that at 18.whatever mhz, micros() is slower. But looking at wiring.c, this does not appear to be the case. That version of micros() is the same as in the official core, which just uses integer division - so the result is fast regardless of clock speed, but wrong (sometimes very wrong) if 64 is not divisible by clock cycles per microsecond. 

It looks like you're referencing a micros() implementation like in my core, where I use a combination of bitshifts and addition/subtraction to get micros() within a few percent of reality, at a modest cost in execution speed (since doing floating point division to get the actual value takes forever).  You're welcome to take my implementation if you want to go that route.

MCUdude, 2018-07-19:
I'd prefer to use the original millis/micros code.

I just tried a 18.432MHz crystal and ran the BlinkWithoutDelay example. It's weird that 1 millis step actually lasts 885 us, while 1000 micros step lasts 1160 us. I'm not quite sure what to write now..

BTW I opened an issue over at the MicroCore repo about millis using timer0. Do you have a trick for creating a good-enough millis function with weird F_CPUs like 9.6, 4.8 and 1.2 MHz (while still also working with 1, 8 and 16 MHz)?

obskyr, 2019-02-21:
So… is the current deal simply "if you use an 18.432 MHz crystal, timing functions will be wildly off", or is there a simple way to get `millis()` and `micros()` to be accurate within a few percent? Do you have to choose between fast UART and being able to time things?

MCUdude, 2019-02-21:
> Is the current deal simply "if you use an 18.432 MHz crystal, timing functions will be wildly off"

Yes. millis and micros doesn't go well with weird clocks.

> Do you have to choose between fast UART and being able to time things?

Well, you don't _have_ to use 115200, 230400, and 460800. Why don't run your AVR at 16 MHz and use 250000, 500000 or 1M?

obskyr, 2019-02-21:
The UART has to be compatible with PCs, and PC serial ports typically only support common baud rates. For example, here are the baud rates my computer supports:

[image]

Below the 115200 baud rate I'm aiming for, the highest applicable baud rate with an error margin of under ±2% is 38400 – that's ⅓ of 115200, which sadly isn't an option for my purposes.

Is there a chance that MiniCore will support sort-of accurate (not *accurate* accurate, but reasonably approximated) `millis()` and `micros()` anytime soon? It'd be very useful for applications where non-critical timing is used (for example, I only really need `millis()` to be accurate to ~10 milliseconds, and `micros()` to ~10 microseconds).

SpenceKonde, 2019-02-22:
If you look at what i did with ATTinyCore, it is possible to implement
millis such that it does a pretty good job of the weird clocks (with
specific combinations of bitshifts and addition for each weird clock speed)
without taking forever to return. You're welcome to use that implementation
in MiniCore (and megacore, mightycore, etc).

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: spencekonde@gmail.com

Jackjan4, 2019-02-22:
@obskyr I'm preparing a pull request for accurate micros() and millis() for 24MHz and 18.432MHz based on @SpenceKonde awesome work.