MCUdude/MiniCore #65: not able to code atmega88PB in arduino ide

bhaveshk023, 2018-07-27:
I have burn the bootloader for atmega88pb and that is sucessfully done. But i am not able to code it using the keyword provided in the data on github. I am pasting my simple test code for led blink.
#define led  PB1 
void setup() {
pinMode(led, OUTPUT);
}
void loop() {
digitalWrite(led, HIGH);
delay(100);
digitalWrite(led, LOW);
delay(100);
}
This is compiling properly, but after uploading the led is not blinking...
Please Help!!

per1234, 2018-07-27:
The Arduino API IO functions take Arduino pin numbers. PB1 is not an Arduino pin number.

If you want to use Arduino functions then you need to use Arduino pin numbers. If you look at the [pinout documentation in the readme](https://github.com/MCUdude/MiniCore#pinout) you'll see that PB1 is Arduino pin 9. So if you make this change to your code
```
#define led 9
```
You'll find PB1 merrily blinking. Your previous code actually happened to blink pin 1 (PD1) because the macro PB1 is defined as 1.

If you want to use the port/bit notation (e.g. `PORTB`, `PB1`, `PINB`) then you'll need to do direct register manipulation rather than using the Arduino functions.

bhaveshk023, 2018-07-27:
ohh i made that changes in my code and it works .. Thank u per123.

closed 2018-07-27 by bhaveshk023