MCUdude/MiniCore #67: PinChange interrupts don't work on D20/D21 (PB6/PB7)

WestfW, 2018-09-19:
minicore supports using PB6/PB7 (normally the XTAL pins) as normal digital IO, and the various digitalPinToPCxxx() macros have been updated to include those, except I don't think they've been fully updated...
I don't see any logic to force them back into PORTB (PCMSK0):

~~~
#define digitalPinToPCICR(p)    (((p) >= 0 && (p) <= 21) ? (&PCICR) : ((uint8_t *)0))
#define digitalPinToPCICRbit(p) (((p) <= 7) ? 2 : (((p) <= 13) ? 0 : 1))
#define digitalPinToPCMSK(p)    (((p) <= 7) ? (&PCMSK2) : (((p) <= 13) ? (&PCMSK0) : (((p) <= 21) ? (&PCMSK1) : ((uint8_t *)0))))
#define digitalPinToPCMSKbit(p) (((p) <= 7) ? (p) : (((p) <= 13) ? ((p) - 8) : ((p) - 14)))
~~~

MCUdude, 2018-09-19:
Thanks for notifying me.
I haven't tested the following code, but this should hopefully fix it.

``` c
#define digitalPinToPCICR(p)    (((p) >= 0 && (p) <= 22) ? (&PCICR) : ((uint8_t *)0))
#define digitalPinToPCICRbit(p) (((p) <= 7) ? 2 : (((p) <= 13) ? 0 : (((p) <= 19) ? 1 : (((p) <= 21) ? 0 : -1))))
#define digitalPinToPCMSK(p)    (((p) <= 7) ? (&PCMSK2) : (((p) <= 13) ? (&PCMSK0) : (((p) <= 19) ? (&PCMSK1) : (((p) <= 21) ? (&PCMSK0) : (((p) <= 22) ? (&PCMSK1) : ((uint8_t *)0))))))
#define digitalPinToPCMSKbit(p) (((p) <= 7) ? (p) : (((p) <= 13) ? ((p) - 8) : (((p) <= 21) ? ((p) - 14) : (((p) <= 22) ? ((p) - 16) : -1))))
```

WestfW, 2018-09-19:
Hmm.   It's getting sort of ugly :-(
Aren't ...ToPCICRbit() and ...ToPCMSKbit() giving the same (or almost the same?) values as the
are in the digital_pin_to_port_PGM[] and digital_pin_to_bit_mask_PGM[] arrays?
And ...ToPCMSK() is just &PCMSK0+...ToPCICRbit()

(I suppose a lot depends on how much you want to diverge from the Arduino branch, style-wise.)

MCUdude, 2018-09-19:
> Aren't ...ToPCICRbit() and ...ToPCMSKbit() giving the same (or almost the same?) values as the
are in the digital_pin_to_port_PGM[] and digital_pin_to_bit_mask_PGM[] arrays?

`digital_pin_to_port_PGM[]` and `digital_pin_to_bit_mask_PGM[]` is there to map digital pin numbers to the physical pins. Nothing to do with PCINTS at all. The output _may be the same as the PCINT related macros, but IMO it's not a good idea to mix these two.

> And ...ToPCMSK() is just &PCMSK0+...ToPCICRbit()

That's correct. However this is only true because they are placed one after the other on the register list. This is not be true for all AVRs (look at [PCMSK3](https://github.com/ExploreEmbedded/Tit-Linux-64/blob/9b6946d2bd42058305b78d0eb85671939f9851da/Tit-Linux-64/tools/packs/atmega328pb/include/iom328pb.h#L515) which BTW is missing from [the datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/40001906A.pdf)). I also think that the code/macro is more readable when the actual register is refered to rather than PCMSK0+offset

commit 1906914f (2018-09-30): Fix PCINT related macro issue
Closes #67
closed 2018-09-30 by MCUdude