MCUdude/MiniCore #68: Optiboot bootloader not responding

macdonaldtomw, 2018-09-21:
Hello,

I'm trying to write an app that will turn an Arduino into a stand-alone STK500 programmer that will flash decode a .hex file stored on a connected SD card, and then flash that hex file to a ATmega328PB that is connected to the Arduino via UART using STK500 protocol (communicating directly with Optiboot bootloader).  See below image:

[image]

Here are the steps I am following:

1) Connect USBTiny ISP to my PC and to my ATmega328PB board via ISP header
2) Select "burn bootloader" command from within Arduino IDE:  _Tools-> Burn Bootloader_  
3) Verify bootloader was successfully flashed to ATmega328PB using the status window in Arduino IDE
3) Disconnect ISP header from my ATmega328PB board
4) Connect my Particle Electron to the ATmega328PB board (only signals are UART_TX, UART_RX... and common ground)
5) Run app on Particle Electron that resets the ATmega328PB using an external watchdog-based reset circuit connected to UART_TX signal (hold UART_TX low for >0.8seconds will pull ATmega328PB reset line LOW), and then immediately attempts to read the signature bytes from the ATmega328PB using the STK500 _READ_SIGN_ command

Nothing happens (I get no response from the ATmega328PB).

Here is a logic analyzer connected to UART signals as well as the reset signal of the ATmega328PB.

Am I doing something wrong?

[image]
[image]

MCUdude, 2018-09-21:
I've never used Optiboot with anything else that a USB to serial adapter, so I don't know what's wrong with your setup.
 I'd start out by uploading a hex file to the ATmega328PB using a USB to serial adapter, and capture the whole session with a logic analyzer. Then I'd switch back to your setup and try again, just to compare the two.

However a logic analyzer doesn't capture "real" voltages. It therefore may be an issue with the threshold voltages.

MCUdude, 2019-03-18:
Did you ever solve this?

macdonaldtomw, 2019-03-18:
I did!

MCUdude, 2019-03-18:
Awesome!

closed 2019-03-18 by MCUdude