MCUdude/MiniCore #72: Killed a bunch of Arduino Pro Mini with the 1Mhz option

slave2anubis, 2018-11-05:
Hi, just found this nice bootloader, and its just what i am needind in my ultra low power sensor application. After using a USBAsp to program a Arduino Pro Mini with the 8Mhz internal clock option, everything worked as expected... the LED (PIN13) did a heart beat kinda blinking.
But i was after the 1Mhz clock option since it, makes sense to get as low clock as posibile, sadly  after burinng the bootloader again using this option the board died, tried on a few other ones the same issue... Working fine on 8Mhz but dying after the 1Mhz flash.
Any hints on what i am doing wrong?

Thanks.

MCUdude, 2018-11-06:
Hi!
I think I know what's wrong. You see the ISP clock frequency of the USBasp is on many cases higher that on other programmers. The rule is that the ISP clock should be a maximum of 1/4 of the clock the microcontroller runs on. My USBasp has an ISP clock of around 730kHz, which is WAY to high to interface with a microcontroller running at 1 MHz.

What you need to do is to add an extra parameter to the AVRdude command `-B[divider]`. I use `-B32` to run the USBasp at 1/32th of the speed. The simplest way of doing this is to install [MicroCore](https://github.com/MCUdude/MicroCore), which moved you a set of "slow" programming options such as _USBasp (slow)_.

[image]

slave2anubis, 2018-11-07:
Thank's man! Been abel to resurect 2 out of the 3 with this method, and get them to work perfectly at 1Mhz internal clock.

closed 2018-11-30 by MCUdude