MCUdude/MiniCore #73: Atmega328pb bootloader not working

ZoomStar, 2018-11-23:
Hi, 

I have tried all of this week to burn a bootloader over my atmega328PB.... but i have failed 8 times ....in a row...

The problem is that i have now 8 "bricks" because after i tried to burn this...the chip is no longer responsiv to my commands ....

i will attach here the log/error from Arduino ide ...

I have only 2 atmega328pb left...please i don't want to brick this too...

THANK YOU IN ADVANCE...
George.
[err.txt](https://github.com/MCUdude/MiniCore/files/2611475/err.txt)

MCUdude, 2018-11-23:
Sorry, I'm not able to look into this at the moment. It _may_ be because the 328PB isn't able to drive the external oscillator because the 328PB can't drive an external oscillator as "hard" as the 328P can.

You may also try [Watterott's repo](https://github.com/watterott/ATmega328PB-Testing).
Also, you should also check if the 1MHz or 8 MHz internal oscillator option work. Note that the USBasp is usually too fast for a 1 MHz clock, so you should probably use 8 MHz internal instead.

MCUdude, 2019-02-12:
You haven't replied to your issue here and over at the Pololu repo. Did you solve the issue or did you just give up?

ZoomStar, 2019-02-12:
Sorry for late response. 

No, i will never give up on anything (about electronics, always there is a way to make it work)

So i removed the 16mhz crystal from my custom board and i have burn the 8mhz bootloader.

And now is working like a charm. 
The only thing that scares me right now is that i pass 23kb of the flash memory with my code, and the Arduino IDE is warning me that is possible to have instability. if i will have any problem with that i will come here again, with a new topic.

Thank you so much for your help!

MCUdude, 2019-02-12:
Great to hear you got it working! I'm pretty sure your issue is caused by the lacking full swing oscillator option on the 328pb. IMO it was a stupid move by Atmel to remove this feature.

closed 2019-02-12 by MCUdude