MCUdude/MiniCore #74: Cannot compile

gcormier, 2018-11-29:
Perhaps similar to #47. I can't compile using MiniCore for a 328PB.

Targeting Arduino Pro - compiles, no errors.

This is using SPIMemory library to interface with WinBond flash chips.

MiniCore:
```
In function 'void loop()':

IRSniffer:117: error: invalid conversion from 'volatile byte* {aka volatile unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]

       flash.readByteArray(dumpy, packet, 8);

\Arduino\libraries\SPIMemory\src/SPIFlash.h:63:12: note:   initializing argument 2 of 'bool SPIFlash::readByteArray(uint32_t, uint8_t*, size_t, bool)'

   bool     readByteArray(uint32_t _addr, uint8_t *data_buffer, size_t bufferSize, bool fastRead = false);

error: invalid conversion from 'volatile byte* {aka volatile unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]

         flash.readByteArray(dumpy, packet, 8);

Documents\Arduino\libraries\SPIMemory\src/SPIFlash.h:63:12: note:   initializing argument 2 of 'bool SPIFlash::readByteArray(uint32_t, uint8_t*, size_t, bool)'

   bool     readByteArray(uint32_t _addr, uint8_t *data_buffer, size_t bufferSize, bool fastRead = false);

exit status 1
invalid conversion from 'volatile byte* {aka volatile unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
```

gcormier, 2018-11-29:
None of the variants will compile (328, 328P)

Library:
https://github.com/Marzogh/SPIMemory

MCUdude, 2018-11-30:
Well, all the SPIflash library examples compiles just fine here.
Have you written the IRSniffer sketch that the error refers to?

The issue is clearly stated here:
```
error: invalid conversion from 'volatile byte* {aka volatile unsigned char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
```

The reason why this compiles when selecting Pro mini or UNO is because the official Arduino core has the -fpermissive flag enabled. I have no idea why they did this because it encourage you to keep your bad coding practice by [reducing actual errors to warnings](https://stackoverflow.com/questions/8843818/what-does-the-fpermissive-flag-do). By default Arduino IDE does not show warnings in the log, only errors

gcormier, 2018-11-30:
Ah, that is very annoying for it to do that. Looks like there's no simple way to enable it either short of editing some configuration files in the Arduino folders.

I'll look to fixing up the actual code.

closed 2018-11-30 by gcormier

eejanuz, 2019-05-01:
I have problems compile with Minicore when wan't to use the Blynk Library https://github.com/blynkkk/blynk-library it throws allot of error.`Arduino: 1.8.9 (Windows 7), Board: "ATmega328, No bootloader, 328P / 328PA, Disabled, LTO enabled, 16 MHz external"

In file included from C:\Users\JS\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.2\cores\MCUdude_corefiles/Arduino.h:316:0,

                 from sketch\MCO_100_Blynk_test.ino.cpp:1:

C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkApiArduino.h: In member function 'void BlynkApi<Proto>::processCmd(const void*, size_t)':

C:\Users\JS\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.2\variants\standard/pins_arduino.h:121:78: error: expected ':' before ')' token

 #define analogInputToDigitalPin(p)  ((p < 5) ? (p) + 14 : ((p < 8) ? (p) + 17) : -1)

                                                                              ^

C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkApiArduino.h:71:62: note: in expansion of macro 'analogInputToDigitalPin'

     #define BLYNK_DECODE_PIN(it) (((it).asStr()[0] == 'A') ? analogInputToDigitalPin(atoi((it).asStr()+1)) : (it).asInt())

                                                              ^

C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkApiArduino.h:94:19: note: in expansion of macro 'BLYNK_DECODE_PIN'

     uint8_t pin = BLYNK_DECODE_PIN(it);

                   ^

C:\Users\JS\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.2\variants\standard/pins_arduino.h:121:78: error: expected primary-expression before ')' token

 #define analogInputToDigitalPin(p)  ((p < 5) ? (p) + 14 : ((p < 8) ? (p) + 17) : -1)

                                                                              ^

C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkApiArduino.h:71:62: note: in expansion of macro 'analogInputToDigitalPin'

     #define BLYNK_DECODE_PIN(it) (((it).asStr()[0] == 'A') ? analogInputToDigitalPin(atoi((it).asStr()+1)) : (it).asInt())

                                                              ^

C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkApiArduino.h:94:19: note: in expansion of macro 'BLYNK_DECODE_PIN'

     uint8_t pin = BLYNK_DECODE_PIN(it);

                   ^

C:\Users\JS\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.2\variants\standard/pins_arduino.h:121:80: error: expected ')' before ':' token

 #define analogInputToDigitalPin(p)  ((p < 5) ? (p) + 14 : ((p < 8) ? (p) + 17) : -1)

                                                                                ^

C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkApiArduino.h:71:62: note: in expansion of macro 'analogInputToDigitalPin'

     #define BLYNK_DECODE_PIN(it) (((it).asStr()[0] == 'A') ? analogInputToDigitalPin(atoi((it).asStr()+1)) : (it).asInt())

                                                              ^

C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkApiArduino.h:94:19: note: in expansion of macro 'BLYNK_DECODE_PIN'

     uint8_t pin = BLYNK_DECODE_PIN(it);

                   ^

In file included from C:\Users\JS\Documents\Arduino\libraries\Blynk\src/Adapters/BlynkEthernet.h:24:0,

                 from C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkSimpleUIPEthernet.h:20,

                 from C:\Users\JS\Documents\Arduino\Chargecontroller\MobilKamera\ver3.1board\MCO100\MCO_100_Blynk_test\MCO_100_Blynk_test.ino:2:

C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkApiArduino.h:94:39: error: expected ':' before ';' token

[... 64 lines cut ...]
                   ^

In file included from C:\Users\JS\Documents\Arduino\libraries\Blynk\src/Adapters/BlynkEthernet.h:24:0,

                 from C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkSimpleUIPEthernet.h:20,

                 from C:\Users\JS\Documents\Arduino\Chargecontroller\MobilKamera\ver3.1board\MCO100\MCO_100_Blynk_test\MCO_100_Blynk_test.ino:2:

C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkApiArduino.h:102:39: error: expected ':' before ';' token

             pin = BLYNK_DECODE_PIN(it);

                                       ^

C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkApiArduino.h:102:39: error: expected primary-expression before ';' token

C:\Users\JS\Documents\Arduino\libraries\Blynk\src/BlynkApiArduino.h:102:39: error: expected ')' before ';' token

exit status 1
Error compiling for board ATmega328.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`