MCUdude/MiniCore #75: convert String to Char not work correctly

Hawkfree, 2018-12-12:
I have program to sending data to server.
**1) use clasic bootloader** - the server receives a correct data
log: 192.168.0.106 - - [12/Dec/2018:09:31:53 +0100] "GET /arduino/elektromer.php?ID=AR06&t1=12&spotreba=12558&tarif=1" 200 2 "-" "-"
**2) use minicore bootloader** - the server NOT receives a correct data
log: 192.168.0.106 - - [12/Dec/2018:09:30:45 +0100] "GET /arduino/elektromer.php?ID=AR06&t\x06\xa8\b\xdc\xa5\x03\x122\x04\xa6\xbe\xca\x0f\x10\xdd\b'\x04l\x03\xba\b\xa5\x03K\x04\x98\x01 HTTP/1.0" 400 981 "-" "-"

`void sendPacket(String dataString){

  char dataChar[50];
  dataString.toCharArray(dataChar, 50);      
  ether.browseUrl(PSTR("/arduino/elektromer.php?"), dataChar, server, callback);

}`

Thank you for your advice.

MCUdude, 2018-12-12:
Can you isolate the problem please (smallest code possible while still being able to demonstrate the problem), and then upload the code so I'm able to test it myself?

Hawkfree, 2018-12-13:
I found the solution:
Convert data in loop{} not before sending packet.

global declaration:
`char dataChar[51];`

`///////////////////////////////////////////////////////////////////////////////////////////////////////
void loop () { 
  ethPaket = ether.packetLoop(ether.packetReceive());
  if(ethPaket) prijemPaketu();
  wdt_reset();         
  if (millis() > timer + 5000) 
  {
    timer = millis();
    power=random(1, 5000);
    t=random(-50, 50);

    dataString = ArduinoID + "&t1=" + t + "&power=" + power + "\r\n";
    dataString.toCharArray(dataChar, 50); 
    
    odeslaniPaketu(dataChar);
  } 
}//end loop`

`//////////////////////////////////////////////////
void odeslaniPaketu(char dataChar[]){
  ether.browseUrl(PSTR("/shome/arduino/elektromer.php?"), dataChar, server, callback);
}`

closed 2019-02-12 by MCUdude