MCUdude/MiniCore #78: delay() on 20Mhz is wrong

Jackjan4, 2019-02-11:
Since MiniCore uses the original wiring.c (I suppose?), delay is off when not using frequencies that are fractions of 64. 20Mhz is obviously not a fraction of 64 and gives wrong delay timings when used.
Simple LED tests can show how timing drafts away compared to 16Mhz (external in the Uno for example) or 8Mhz internal.

The problem resides in the return statement of delay() which returns an int, however the division result is a fraction.

MCUdude, 2019-02-12:
I am aware if the issue. All my Arduino cores uses standard wiring.c. Feel free to issue a pull request [here](https://github.com/MCUdude/MCUdude_corefiles) if you're able to come up with something better for 20 and 24 MHz

Jackjan4, 2019-02-16:
Since this is a topic for the corefiles then I will close this and open an issue [there](https://github.com/MCUdude/MCUdude_corefiles/issues/13).

closed 2019-02-16 by Jackjan4