MCUdude/MiniCore #79: Make baud rate an option in IDE

Jackjan4, 2019-02-11:
This is more a suggestion:

Since ceramic resonators on high frequencies (20 & 24Mhz) are very picky when it comes to uploading-baud-rate, I'm always experimenting in boards.txt to find the correct baud rate. Wouldn't it be a good option to enable to set baud-rates in the IDE as a new menu? 
(This probably needs the function that baud rate is properly selected by default, depending on clock frequency, does Arduino IDE support dependencies in menus?)

MCUdude, 2019-02-12:
First, it would add another menu option to the tools menu. I've decided not to "expose" the baudrate because most users doesn't need to change it anyways. If you're having troble with the clock source it may be a good idea to use an external clock with a "driven" output you can feed straight into the XTAL1 pin on your AVR.

Second, it would be kinda hard to implement it in an elegant way. The problem is the way the boards.txt file is written and interperated by the IDE. Ideally we want the working baud rates to be available in a separate sub menu when a spesific clock frequency is selected. The problem is that it's not possible to change the content in one sub menu based on the selected option in another submenu. 

> does Arduino IDE support dependencies in menus?
AFAIK, no 😢

Jackjan4, 2019-02-12:
> First, it would add another menu option to the tools menu. I've decided not to "expose" the baudrate because most users doesn't need to change it anyways. If you're having troble with the clock [...]

Okay, that confirms my guess. Too bad that boards.txt allows such things. Thank you very much. Closing this now.

closed 2019-02-12 by Jackjan4