MCUdude/MiniCore #81: Travis CI fails

MCUdude, 2019-02-18:
The last boards.txt commit cause CI to fail. Care to look into it @per1234 ?

MCUdude, 2019-02-18:
It seems like it's failing because it builds with `bootloader:true`. Do we really need to do separate builds for targets with and without bootloaders? We don't do this with MightyCore and MegaCore.

per1234, 2019-02-18:
You're correct about the problem. With some versions of the Arduino IDE, it is possible to leave out menu options from the FQBN and the IDE will just use the default option, but that doesn't work for all IDE versions. I think we should cover all the bootloader menu options in the test anyway. It won't add any extra time to the build because I can do it within the existing tests. I'll submit a pull request.

per1234, 2019-02-19:
I think I have the issue resolved, but the Travis CI build still fails due to these warnings (because arduino-ci-script's `set_board_testing` is enabled):
https://travis-ci.org/per1234/MiniCore/builds/494815192
```
Bootloader file specified but missing: /home/travis/Arduino/hardware/MiniCore/avr/bootloaders/optiboot_flash/bootloaders/atmega328p/16000000L/optiboot_flash_atmega328p_UART0_115200_16000000L.hex
```
It seems that the `bootloaders` folder was added to the optiboot_flash repo, but that the subtree has not yet been updated in MiniCore, yet the `bootloader.file` definitions were updated in boards.txt in anticipation of this, resulting in the "Bootloader file specified but missing" warnings.

Would you like me to submit a pull request for the update to the Travis CI configuration to reflect the changes in the bootloader menu now, or do you want me to wait until it will result in a passing Travis CI build?

MCUdude, 2019-03-29:
Have you come up with a solution yet @per1234? 🙂

per1234, 2019-03-30:
I came up with a solution 1.5 months ago when I made my previous comment. I have been waiting for you to answer my question I asked in that comment. Although you never did do that, I see you have since resolved the issue via https://github.com/MCUdude/MiniCore/commit/92536a70dd64fc49974b4c30dcd576bf32be7246. Ironically, this was done in response to https://github.com/MCUdude/MiniCore/issues/83, rather than my report of the same issue that came a month before that issue was opened.

MCUdude, 2019-03-30:
I'm sorry, I have no idea (and no excuse really) why I didn't respond to your previous reply. Like you pointed out I didn't even solve the issues you pointed out before 92536a7. Would you like to submit a PR with the changes you've made?

per1234, 2019-03-30:
I rebased my branch on the latest version of MiniCore and am running a fresh Travis CI build to make sure it's all good. Once that's done, I'll submit the PR.

closed 2019-03-30 by MCUdude