MCUdude/MiniCore #83: boards.txt has incorrect path to bootloaders

SpenceKonde, 2019-03-17:
As of cd448f0b7200a9b707e57e44d361b0fa02515f10 the path to the bootloaders in boards.txt does not match the file layout. 

The path was changed from 
optiboot_flash/{build.mcu}/{build.f_cpu}/optiboot_flash_{build.mcu}_{upload.port}_{upload.speed}_{build.f_cpu}.hex

to

optiboot_flash/bootloaders/{build.mcu}/{build.f_cpu}/optiboot_flash_{build.mcu}_{upload.port}_{upload.speed}_{build.f_cpu}.hex

Note the additional "bootloader" directory in the path, which does not match the locations of the files.

MCUdude, 2019-03-17:
I can't believe how I missed this! Arduino IDE even highlights this warning when compiling for a MiniCore target! Should be fixed in 92536a7. Turned out I forgot to update the optiboot flash subtree.

closed 2019-03-17 by MCUdude

MCUdude, 2019-03-17:
BTW how do _you_ build boot loaders on a Windows computer? I'd like to update the optiboot_flash makefile to automatically use the avr-gcc that comes with Arduino IDE if no custom path or command is already specified. I'd like this to work on both Windows and macOS, and preferably if the makefile itself can detect what OS is used and take actions based on this. The makefile that's used at the moment have references to Arduino 1.0.X and 1.5.X and you'll have to [uncomment](https://github.com/MCUdude/optiboot_flash/blob/959494fa1e1e8ac49d976fa1d9c475c2cfd13e9c/Makefile#L40-L43) one of these lines.

SpenceKonde, 2019-03-17:
>.> I use an old 1.0.6 installation for the build tools and manually update it's compiler

MCUdude, 2019-03-17:
Ah, that sounds like a pain in the butt. 

According to this [Stack overflow answer](https://stackoverflow.com/a/54086635) I can install make through choco and use it with cmd.exe. Then I can just use the toolchain that ships with the current install of Arduino IDE.

SpenceKonde, 2019-03-18:
Eh, it's not bad. They don't update the compiler used with the IDE very often, so *shrug*