MCUdude/MiniCore #84: Cannot upload program when running on 3.3V

jpraus, 2019-03-18:
Hi, 

First of all, thank you for such a great project and awesome documentation!

I am running my ATmega8 at 3.3V (because the board itself will be 3.3V). I am able to burn the bootloader on it via USBasp programmer. But I am not able to flash it via UART interface using FTDI programmer afterward. If I run it on 5V everything works like a charm. Using 8MHz internal oscillator. Can this be related to bootloader or board settings in MiniCore library or is it unrelated?

Thank you for any advise!

MCUdude, 2019-03-18:
Great to hear you find MiniCore useful!

This is most likely caused by the internal RC oscillator. This is usually a tiny bit slower when the voltage is lower. What you may try to do is to change the baud rate. Try changing [this line](https://github.com/MCUdude/MiniCore/blob/92536a70dd64fc49974b4c30dcd576bf32be7246/avr/boards.txt#L140) to 250000 instead. A baud rate of 250000 is ideal for 8 MHz because it divides evenly (8000000/250000 is an integer number) and therefore have 0% error.

Note that you'll have to restart Arduino IDE and re-burn the bootloader in order for the changes to take effect.

jpraus, 2019-03-18:
Awesome! On it right away.

jpraus, 2019-03-18:
Sadly, the result is still the same. I can go as low as 3.8V to be able to program it via UART and bootloader. But that is it. I also tried 1MHz internal oscillator but the result was the same.

MCUdude, 2019-03-18:
Sorry, I thought you had an Atmega 328P. Are you sure you edited the correct line in the boards.txt?

jpraus, 2019-03-18:
Yes, I figured that out :) It gave me using non-standard baud rate warning and I saw the 250000 in the console.

MCUdude, 2019-03-18:
So, you're now able to upload when the ATmega8 is running at 3.3V now (with a baud rate of 250000)?

jpraus, 2019-03-19:
No, I am not. And not even at 1MHz (with a baud rate of 9600). It works with voltage 3.8V and above, but below that, it does not. I will try to use different baud rates today at 8MHz to see if it makes any difference.

MCUdude, 2019-03-19:
Are you using an ATmega8, 8L or 8A? According to the datasheet the atmega8 is rated from 4.5V to 5.5V.

jpraus, 2019-03-19:
Atmega8A

MCUdude, 2019-03-19:
Well, I'm not sure how much more help I can provide then. It is a known fact the internal oscillator may be slower at 3.3V than on 5V, and UART requires an accurate clock. I'd probably just use an external 8 MHz oscillator instead.

jpraus, 2019-03-19:
No worries, you helped me a lot! At least now I know the possible cause. An external oscillator is an unwanted complication to the circuit, but I will think of that. I will try to use an external oscillator to see if it makes the difference.

MCUdude, 2019-03-19:
I have messed up on this before too.
A while ago I designed a small touch panel based on the ATmega88A. To keep the cost down I decided to stick with the internal oscillator, even though the touch panel should be able to communicate over RS485. It turned out the oscillator was very inaccurate, and every panel needed oscillator calibration in order to function properly. Because out batch programmer couldn't perform OSCCAL I ended up redesigning the board and adding a small external oscillator instead. If you can handle SMD components and are tight on space I can recommend the Murata CSTNE8M00G55.

jpraus, 2019-03-19:
Yes, it's resolved. The internal oscillator is the problem. With an external one, it works like a charm. Thank you so much for your help! Now I need to decide whether to introduce oscillator to the board or go with 5V.

closed 2019-03-19 by jpraus