MCUdude/MiniCore #86: avr/bootloaders/optiboot_flash/makeall file has leftover git merge snippets

zgyarmati, 2019-04-05:
Introduced with: 92536a70dd64fc49974b4c30dcd576bf32be7246 , see 
https://github.com/MCUdude/MiniCore/blame/master/avr/bootloaders/optiboot_flash/makeall

MCUdude, 2019-04-05:
Thanks for reporting! I wasn't aware of this. @per1234 do you know an easy way to fix this?
When trying to update the optiboot_flash subtree I'm getting an this error:

```
$ git subtree pull --prefix avr/bootloaders/optiboot_flash https://github.com/MCUdude/optiboot_flash.git master --squash
From https://github.com/MCUdude/optiboot_flash
 * branch              master     -> FETCH_HEAD
Auto-merging avr/bootloaders/optiboot_flash/README.md
Auto-merging avr/bootloaders/optiboot_flash/Makefile
CONFLICT (content): Merge conflict in avr/bootloaders/optiboot_flash/Makefile
Automatic merge failed; fix conflicts and then commit the result.
```

MCUdude, 2019-04-06:
It should be OK now. Thanks for reporting!

closed 2019-04-06 by MCUdude