MCUdude/MiniCore #88: Bootloader upload issue in ATmega8 at '16 MHz external' clock option.

manj9501, 2019-04-16:
Whenever I try to upload bootloader on an ATmega8 plugged into an Arduino UNO, with the '16 MHz external' clock selected, I get the following error log:
```
avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\YoYo\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.2/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega8
         Chip Erase delay              : 10000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20   128    0 no        512    4      0  9000  9000 0xff 0xff
           flash         33    10    64    0 yes      8192   64    128  4500  4500 0xff 0x00
           lfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           efuse          0     0     0    0 no          0    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  2000  2000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9307 (probably m8)
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Error while burning bootloader.
Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified
avrdude: reading input file "{bootloader.extended_fuses}"
avrdude: writing efuse (0 bytes):

Writing | ################################################## | 100% -0.00s

avrdude: 0 bytes of efuse written
avrdude: verifying efuse memory against {bootloader.extended_fuses}:
avrdude: load data efuse data from input file {bootloader.extended_fuses}:
avrdude: input file {bootloader.extended_fuses} contains 0 bytes
avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 0 bytes of efuse verified
avrdude: reading input file "0b110{bootloader.ckopt_bit}0100"
avrdude: invalid byte value (0b110{bootloader.ckopt_bit}0100) specified for immediate mode
avrdude: read from file '0b110{bootloader.ckopt_bit}0100' failed

avrdude done.  Thank you.
```
Notably, I can successfully upload the bootloader on an ATmega8 when I select '8 MHz internal' clock option, with the same set-up. Also I have no problems whatsoever, burning the MiniCore bootloader to an ATmega328P, using the same set-up, and the '16MHz external' clock option.
Following is a screenshot showing settings applied when I get the error with ATmega8:

[image]

MCUdude, 2019-04-16:
Thank you for reporting! I'm pretty sure it's caused by [this line here](https://github.com/MCUdude/MiniCore/blob/d58ede1bae57954596faa34659f2651f2b986ea0/avr/boards.txt#L511). Can you please try to replace 20MHz to 16MHz and report back?

manj9501, 2019-04-16:
I edited the "...\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.2\boards.txt" file at line 511 as you instructed, and lo and behold, issue resolved! Thanks a lot for the prompt response!

closed 2019-04-16 by MCUdude

ItsZelix, 2022-07-23:
Hey! I'm facing the same issue, but with an 8MHz internal clock. Could you please share the boards.txt file you were using for this?

manj9501, 2022-07-25:
Hi @ItsZelix 
I so wish I could help you out but I've changed my laptop and consequently, I no longer have the necessary boards.txt in my possession. My apologies!
I hope you find some alternative solution soon!
Or maybe open a new issue.

commit 663e8a55 (2019-04-16): Fix typo
Closes #88