MCUdude/MiniCore #89: Burn without hardware programmer?

PRosenb, 2019-04-19:
Hi guys,
It would be great to be able to burn MiniCore without needing to own a hardware programmer. @nickgammon uses such an approach with a second Arduino.
https://github.com/nickgammon/arduino_sketches
Possibly the same principle could be used for MiniCore?

MCUdude, 2019-04-19:
If you don't want to buy a dedicated hardware programmer you can always just use an Arduino as ISP.

The problem with Nick Gammons programmer is that it only holds one single hex file for all 328 options. It also spoofs the device signature, turning all 328 targets (328, 328P and 328PB) into a device with a 328P device signature. Fuses are also hardcoded based on the detected target

MiniCore is the exact opposite. It has a separate bootloader file for every chip type, variant, clock frequency and baud rate. It never spoofs any device signatures, and that's why you'll have to select your current variant in the tools menu when you're using MiniCore.

It would require a good amount of work in order to make Gammons HW programmer to be as versatile as the MiniCore tools menu. If Nick himself wants to do this, that's great! Me on the other side, I couldn't be bothered really. You should probably open this issue over at his repository.

closed 2019-04-19 by MCUdude

PRosenb, 2019-04-19:
Many thanks @MCUdude for your quick reply.
I see, makes sense.
I tried to use an Arduino as ISP but couldn't get it to work. Have you had luck with it?

It's not that I don't want to buy a hardware programmer but I think that it's a hurdle for others. It would help to spread when there would be a good manual without additional hardware requirements.

nickgammon, 2019-04-19:
My programmer doesn't change (spoof) the device signature. I don't know how to do that or if it is possible.

MCUdude, 2019-04-19:
I may have been a bit unclear. When you burn a "328P signed" bootloader into an ATmega328 (non P) or 328PB, it will identify itself as a 328P when communicating with it through UART.

nickgammon, 2019-04-19:
How, out of curiosity? The communication between the 328P and the UART (by which I assume you mean the USB interface chip, like the Atmega16U2), it is purely via the serial pins D0 and D1. There is no provision there for a signature to be sent to the UART.

I agree that in the case of the bootloader for the Leonardo (which has the same chip as the Micro) then the bootloader identifies itself as one or the other (I don't remember which) when USB enumeration takes place.

MCUdude, 2019-04-19:
> How, out of curiosity? The communication between the 328P and the UART (by which I assume you mean the USB interface chip, like the Atmega16U2), it is purely via the serial pins D0 and D1. There is no provision there for a signature to be sent to the UART.

There are some old threads on the forum where some users have accidentally ordered ATmega328's instead of ATmega328P's. What most users used to end up with was to shoe-horn the UNO (328P) bootloader into the 328 by ignoring the current device signature (adding the -F flag to the Avrdude command. Now, when communicating with the chip (using your favorite USB to UART chip) it identifies itself as an ATmega328P since the bootloader was built for a 328P and not a 328.
[The bootloader holds the actual signature](https://github.com/MCUdude/optiboot_flash/blob/d50bf930e722102c589737dcceb890f6fedab1c4/optiboot_flash.c#L813-L819), SIGNATURE0, 1 and 2 is pulled from iom328p.h (or whatever the build target is). Therefore it is not read out of the hardware anymore.

This isn't really that big of a problem since the 328 and 328P are binary compatible. It kinda messes up the logic though (a 328 chip identifies itself as a 328P). The 328PB, on the other hand, are different and. "328P" code may work (not considered safe), but if you're building for a 328P and uploading it to a 328PB you'll never be able to take advantage of the additional timers and peripherals the 328PB has.

nickgammon, 2019-04-22:
I think you are confusing two different behaviours here. The bootloader's job is to upload code to the chip. Providing the bootloader actually runs then it will do that job. As I mention [here](https://arduino.stackexchange.com/questions/15936/what-happens-when-code-is-uploaded-using-the-bootloader) Optiboot returns the signature for the Atmega328P without actually reading the hardware signature. I can understand your wanting to change that signature in Optiboot (or whatever you are uploading) so that the chip correctly identifies itself. I am presuming that Optiboot did it that way to squeeze into 512 bytes, so they skipped finding the actual signature.

However identifying itself does not affect what code is uploaded by the bootloader. Let's say you compile code for the 328PB, and upload it. That code will run, regardless of what bootloader you use, providing the bootloader actually uploaded it. The compiler, which runs before the uploading step, will compile for whatever architecture **you tell it you have**. The compiler does not check the chip signature.

> What most users used to end up with was to shoe-horn the UNO (328P) bootloader into the 328 by ignoring the current device signature (adding the -F flag to the Avrdude command.

I think they could have simply duplicated some lines in the boards.txt file. That is what tells avrdude what signature to expect. You could copy/paste the entries for the 328P and change the signature to suit the other, almost identical, chip. The extra features you speak of won't affect the uploading process.

> MiniCore is the exact opposite. It has a separate bootloader file for every chip type, variant ...

What you could do is read the **actual signature** in the bootloader, then you truthfully report what chip it is sitting on. You don't need a separate file for each chip type (unless space is an issue which it might be).

MCUdude, 2019-04-22:
> However identifying itself does not affect what code is uploaded by the bootloader. Let's say you compile code for the 328PB, and upload it. That code will run, regardless of what bootloader you [...]

No. Compiling and uploading are two completely separate processes. The problem is Arduino IDE. when you select "Arduino UNO" for instance (which has a 328P chip) you tell both the compiler and avrdude that they're dealing with a 328P. This is fine and straight forward as long as the chip reports its actual/correct device signature.

The problems begin when we are using enhanced hardware like the 328PB is. If we upload the "UNO bootloader" to the 328PB it will identify itself as a 328P. We can now (using Arduino IDE with Arduino UNO selected) treat the 328PB as an Arduino UNO. However, there's no way of using the additional functionality of the 328PB without applying some ugly hacks.

A different option is to modify boards.txt and add 328PB as a separate option. Then we upload the already existing "UNO bootloader" to the 328PB. Now the compiler "knows" we're working with a 328PB. The problem begins when we want to upload our code. Our 328PB identifies itself as a 328P! Since there's no easy way of switching between targets without having to recompile we're pretty much stuck with adding -F to our avrdude command. An alternative is to add a separate line in boards.txt (and modifying platform.txt) to tell avrdude that it should expect a 328P). IMO neither are good alternatives.

It would indeed be better if Optiboot reported the actual signature rather than the signature the bootloader was compiled for. I'm pretty sure this was done to save precious flash memory.