MCUdude/MiniCore #90: Cannot compile

eejanuz, 2019-05-01:
I cannot compile when use the Blynk library.
https://github.com/blynkkk/blynk-library

`Arduino: 1.8.9 (Windows 7), Board: "ATmega328, No bootloader, 328P / 328PA, Disabled, LTO enabled, 16 MHz external"

In file included from \MiniCore\hardware\avr\2.0.2\cores\MCUdude_corefiles/Arduino.h:316:0,

from sketch\MCO_100_Blynk_test.ino.cpp:1:

BlynkApiArduino.h: In member function 'void BlynkApi<Proto>::processCmd(const void*, size_t)':

\MiniCore\hardware\avr\2.0.2\variants\standard/pins_arduino.h:121:78: error: expected ':' before ')' token

 #define analogInputToDigitalPin(p)  ((p < 5) ? (p) + 14 : ((p < 8) ? (p) + 17) : -1)                 
BlynkApiArduino.h:71:62: note: in expansion of macro 'analogInputToDigitalPin'   #define BLYNK_DECODE_PIN(it) (((it).asStr()[0] == 'A') ? analogInputToDigitalPin(atoi((it).asStr()+1)) : (it).asInt())                                                         
BlynkApiArduino.h:94:19: note: in expansion of macro 'BLYNK_DECODE_PIN uint8_t pin BLYNK_DECODE_PIN(it);

                   ^

\MiniCore\hardware\avr\2.0.2\variants\standard/pins_arduino.h:121:78: error: expected primary-expression before ')' token

 #define analogInputToDigitalPin(p)  ((p < 5) ? (p) + 14 : ((p < 8) ? (p) + 17) : -1)

                                                                              ^

BlynkApiArduino.h:71:62: note: in expansion of macro 'analogInputToDigitalPin'

     #define BLYNK_DECODE_PIN(it) (((it).asStr()[0] == 'A') ? analogInputToDigitalPin(atoi((it).asStr()+1)) : (it).asInt())

                                                              ^

BlynkApiArduino.h:94:19: note: in expansion of macro 'BLYNK_DECODE_PIN'

     uint8_t pin = BLYNK_DECODE_PIN(it);

                   ^

\MiniCore\hardware\avr\2.0.2\variants\standard/pins_arduino.h:121:80: error: expected ')' before ':' token

 #define analogInputToDigitalPin(p)  ((p < 5) ? (p) + 14 : ((p < 8) ? (p) + 17) : -1)

                                                                                ^

BlynkApiArduino.h:71:62: note: in expansion of macro 'analogInputToDigitalPin'

     #define BLYNK_DECODE_PIN(it) (((it).asStr()[0] == 'A') ? analogInputToDigitalPin(atoi((it).asStr()+1)) : (it).asInt())

                                                              ^

BlynkApiArduino.h:94:19: note: in expansion of macro 'BLYNK_DECODE_PIN'

     uint8_t pin = BLYNK_DECODE_PIN(it);

                   ^

In file included from Adapters/BlynkEthernet.h:24:0,

                 from BlynkSimpleUIPEthernet.h:20,

                 from C:\Users\JS\Documents\Arduino\Chargecontroller\MobilKamera\ver3.1board\MCO100\MCO_100_Blynk_test\MCO_100_Blynk_test.ino:2:

BlynkApiArduino.h:94:39: error: expected ':' before ';' token

     uint8_t pin = BLYNK_DECODE_PIN(it);

                                       ^

BlynkApiArduino.h:94:39: error: expected primary-expression before ';' token

BlynkApiArduino.h:94:39: error: expected ')' before ';' token

In file included from \MiniCore\hardware\avr\2.0.2\cores\MCUdude_corefiles/Arduino.h:316:0,

                 from sketch\MCO_100_Blynk_test.ino.cpp:1:

\MiniCore\hardware\avr\2.0.2\variants\standard/pins_arduino.h:121:78: error: expected ':' before ')' token

 #define analogInputToDigitalPin(p)  ((p < 5) ? (p) + 14 : ((p < 8) ? (p) + 17) : -1)

                                                                              ^

BlynkApiArduino.h:71:62: note: in expansion of macro 'analogInputToDigitalPin'

     #define BLYNK_DECODE_PIN(it) (((it).asStr()[0] == 'A') ? analogInputToDigitalPin(atoi((it).asStr()+1)) : (it).asInt())

                                                              ^

BlynkApiArduino.h:102:19: note: in expansion of macro 'BLYNK_DECODE_PIN'

             pin = BLYNK_DECODE_PIN(it);

                   ^

\MiniCore\hardware\avr\2.0.2\variants\standard/pins_arduino.h:121:78: error: expected primary-expression before ')' token

 #define analogInputToDigitalPin(p)  ((p < 5) ? (p) + 14 : ((p < 8) ? (p) + 17) : -1)

                                                                              ^

BlynkApiArduino.h:71:62: note: in expansion of macro 'analogInputToDigitalPin'

     #define BLYNK_DECODE_PIN(it) (((it).asStr()[0] == 'A') ? analogInputToDigitalPin(atoi((it).asStr()+1)) : (it).asInt())

                                                              ^

BlynkApiArduino.h:102:19: note: in expansion of macro 'BLYNK_DECODE_PIN'

             pin = BLYNK_DECODE_PIN(it);

                   ^

\MiniCore\hardware\avr\2.0.2\variants\standard/pins_arduino.h:121:80: error: expected ')' before ':' token

 #define analogInputToDigitalPin(p)  ((p < 5) ? (p) + 14 : ((p < 8) ? (p) + 17) : -1)

                                                                                ^

BlynkApiArduino.h:71:62: note: in expansion of macro 'analogInputToDigitalPin'

     #define BLYNK_DECODE_PIN(it) (((it).asStr()[0] == 'A') ? analogInputToDigitalPin(atoi((it).asStr()+1)) : (it).asInt())

                                                              ^

BlynkApiArduino.h:102:19: note: in expansion of macro 'BLYNK_DECODE_PIN'

             pin = BLYNK_DECODE_PIN(it);

                   ^

In file included from Adapters/BlynkEthernet.h:24:0,

                 from BlynkSimpleUIPEthernet.h:20,

                 from C:\Users\JS\Documents\Arduino\Chargecontroller\MobilKamera\ver3.1board\MCO100\MCO_100_Blynk_test\MCO_100_Blynk_test.ino:2:

BlynkApiArduino.h:102:39: error: expected ':' before ';' token

             pin = BLYNK_DECODE_PIN(it);

                                       ^

BlynkApiArduino.h:102:39: error: expected primary-expression before ';' token

BlynkApiArduino.h:102:39: error: expected ')' before ';' token

exit status 1
Error compiling for board ATmega328.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

MCUdude, 2019-05-01:
Thanks for reporting! It should be fixed now. However, you'll have to do a manual install of MiniCore to get the latest fix. Or, you can wait for v2.0.3 to be released

closed 2019-05-01 by MCUdude

eejanuz, 2019-05-01:
Thanks very much, for this great support:)

commit 9b9bef48 (2019-05-01): Fix missing parentheses
Closes #90