MCUdude/MiniCore #92: Boot loop with Watchdog on 168PA (cold boot)

kalsan, 2019-05-06:
First, thanks a lot for providing this sweet bootloader. I found your website after trouble-shooting an Arduino Nano with an ATMEGA168PA chip, which enters a boot loop after a reset caused by the watchdog. The problem is well-known and due to the standard bootloader. To circumvent this, I burned the MiniCore bootloader for 168P/PA and the bootloop seemed gone. Until I unplugged the device and re-plugged it. Turns out:

  - If the Arduino was reset at least once (by a programmer, serial monitor or the reset button) before the watchdog kills it, everything works fine
  - However if the watchdog kills the Arduino shortly after it boots for the first time after a power loss (I tested with 2100ms delay before watchdog enable and with 1s watchdog), it enters the boot loop we know from other bootloaders. However, with MiniCore and unlike the other boot loader, the boot loop can be interrupted by pressing the RESET button. From then on the Watchdog works nicely until the next "cold boot" (removing power supply and plugging it back in).

I'm happy to run tests and give further information you may need for debugging this.

MCUdude, 2019-05-06:
Can you prove some example code, please? Nothing fancy, just a LED blink sketch with WDT.

kalsan, 2019-05-06:
Thanks for the reply and the very informative edit!

Here's the problematic code (I found this online and slightly adjusted some details):

```
#include <Arduino.h>
#include <avr/wdt.h>

#define led 13 // pin 13 is connected to a led on Arduino Uno

void setup()
{
    wdt_disable();
    pinMode(led, OUTPUT);    // set pin mode
    digitalWrite(led, HIGH); // turn the LED on (HIGH is the voltage level)
    delay(2000);              // wait for a tenth of a second
    digitalWrite(led, LOW);  // turn the LED off by making the voltage LOW
    delay(100);              // wait for a tenth of a second

    wdt_enable(WDTO_1S); // enable the watchdog
                           // will fire after 1s without reset
}

void loop()
{
    while (1)
    {
            // wdt_reset();          // uncomment to avoid reboot
    }
}
```

Since I'm using the watchdog to prevent a freeze that occasionally happens immediately after booting, this is bad news, especially since upstream appears to have knowledge of the existance of the bug (but not of how to fix it). Do you happen to have a suggestion for a workaround, by chance?

MCUdude, 2019-05-06:
> Do you happen to have a suggestion for a workaround, by chance?

The comment in the code suggests clearing MCUCR very early in your code. Try adding 
``` c
MCUCR = 0;
```

Another alternative is to ditch the bootloader and upload using a dedicated programmer.

kalsan, 2019-05-06:
**Attempt 1**

By putting `MCUCR = 0;` as the first line in `setup`, nothing changes

**Attempt 2**

Nothing changes if I try to apply the last example by copying it before `setup` (which I do not fully understand!):

```
ISR(WDT_vect)
{
        // must be rebooted
        // configure
        MCUSR = 0;                          // reset flags
       
                                            // Put timer in reset-only mode:
        WDTCSR |= 0b00011000;               // Enter config mode.
        WDTCSR =  0b00001000 | 0b000000;    // clr WDIE (interrupt enable...7th from left)
                                            // set WDE (reset enable...4th from left), and set delay interval
                                            // reset system in 16 ms...
                                            // unless wdt_disable() in loop() is reached first
                                       
        // reboot
        while(1);
}
```

**Attempt 3**

Uploading the full example 3 (sleep and watchdog) does not result in a reset at all.

**No bootloader attempt**

By setting the `bootloader` option to `no` and then burning the bootloader, the device seems to clear the bootloader since I cannot upload sketches any longer. I then upload the sketch from my earlier post using another Arduino as a programmer. The result is the same as with the bootloader the device came with: it enters a bootloop and does not recover when pressing the reset button.

**Attempt to find another bootloader**

Last thing I tried to find the official Arduino bootloader for the chip. Turns out there is no option for the 168P, only 168.

MCUdude, 2019-05-06:
> Last thing I tried to find the official Arduino bootloader for the chip. Turns out there is no option for the 168P, only 168.

You can safely burn the 168 bootloader!

@westfw is there a quick fix for this issue? TBH I've never really written any code that relies on watchdog reset to work.

kalsan, 2019-05-06:
> You can safely burn the 168 bootloader!

The device signature does not match and using -F to override did not appear to work ("not in sync" when uploading sketches later)

MCUdude, 2019-05-06:
Did you select Arduino Pro Mini or Arduino Duemilanove? They don't use Optiboot, but ATMEGABOOT instead. Try turning on verbose upload in the IDE settings to see the complete error message.

kalsan, 2019-05-06:
Thanks! I'll try it tomorrow and let you know.

WestfW, 2019-05-07:
```
ISR(WDT_vect)
{
        // must be rebooted
        // configure
        MCUSR = 0;                          // reset flags
```

The `MCUSR=0;` statement needs to be in setup(), not in the ISR code.

What happens is that MCUSR is "cumulative."  The bootloader (Optiboot latest) doesn't clear it any more, so that the application can access the reset reason.  After power-on, it will say "restarted by poweron" and the application will be started immediately ("fast start".)  If the application doesn't clear MCUSR and a watchdog reset occurs, the bootloader will STILL do a "fast start" because the "reset by poweron" bit is still set.

This is actually documented in the source code:
https://github.com/Optiboot/optiboot/blob/master/optiboot/bootloaders/optiboot/optiboot.c#L572

kalsan, 2019-05-07:
@MCUdude when attempting the option "Arduino Pro or Pro Mini" and circumventing the signature check using `-F`, the bootloader installs fine but the upload results in responses 0x00 and then 0x1c, causing a not-in-sync error (with connection through the programmer, or only 0x00 when connected directly.

However, I got a step further when trying the "Duemillanove or Diecimila" board family: By forcing the bootloader burn and then forcing the upload, I managed to get the scetch to upload once through the programmer. The result was a non-recoverable bootloop as with the original firmware. However I cannot reproduce this experiment, all I get now are not-in-sync messages.

I'll reply to the newer comment in a separate comment.

kalsan, 2019-05-07:
@WestfW thanks for your input. @MCUdude already found this line and made an appropriate recommendation. However, I mistakenly set `MCUCR` (wrong) instead of `MCUSR` (correct) to 0.

So here's the updated code, which works correctly for both "cold" and "warm boot", using the MiniCore bootloader for 168P:

```
#include <Arduino.h>
#include <avr/wdt.h>

#define led 13 // pin 13 is connected to a led on Arduino Uno

void setup()
{
    MCUSR = 0;
    wdt_disable();
    pinMode(led, OUTPUT);    // set pin mode
    digitalWrite(led, HIGH); // turn the LED on (HIGH is the voltage level)
    delay(2000);              // wait for a tenth of a second
    digitalWrite(led, LOW);  // turn the LED off by making the voltage LOW
    delay(100);              // wait for a tenth of a second

    wdt_enable(WDTO_1S); // enable the watchdog
                           // will fire after 1s without reset
}

void loop()
{
    while (1)
    {
            // wdt_reset();          // uncomment to avoid reboot
    }
}
```

Problem solved and learned a lot. Thank you both for your awesome support :-)

closed 2019-05-07 by kalsan

kalsan, 2019-05-07:
Note for other readers: the described fix does **not** work on Arduino Nano with the old bootloader. It requires Optiboot. This appears to be the only option for Nano 168P devices.

Note for Nano 328P devices: Since the Nano 328P has a new bootloader, I'll try burning that one on such a device next. Also I heared that the Nano 328P seems to work with the Uno bootloader as well. Since the watchdog works fine on the latter, it might be worth a shot as well.

kalsan, 2019-05-07:
UPDATE: Success on Nano 328P with the following procedure:

1. Connect the device to the programmer or Arduino via ICSP pins (for the Nano 168P, I had to use the pins D10-D13 and the reset pin, but this does not appear to work on the 328P!)
2. In Arduino IDE, select the official Arduino Nano board 328P (new bootloader, otherwise the watchdog will not work!).
3. Unplug the programmer and plug the USB
4. Upload any sketch. Attention, in PlatformIO, the `platform` must now be set to `nanoatmega328new`

**Why the official Nano bootloader and not Optiboot on the 328P?**

- PlatformIO: using the official new Nano bootloader instead of Optiboot has the advantage that the device can still be programmed using PlatformIO (MiniCore does not support PlatformIO yet, see #25 ).
- Code compatibility: using the official bootloader allows to use "regular" watchdog code found online, without the need for `MCUSR = 0; wdt_disable();`

Note that Optiboot is still required for 168P devices. Also, for users not working with PlatformIO, Optiboot might still be the preferred choice, also for the 328P.

WestfW, 2019-05-07:
There are essentially four generations of bootloaders:
1) Original ATmegaBoot:  You can't use the watchdog with this - the bootloader has about a 10s timeout, and the max WDT is 8s (and it switches back to 16ms at reset.)  So if you enable the WDT in an application and it fires, you will be stuck with the WDT continually firing in the bootloader.
2) ATmegaBoot with WDT fix: this resets MCUSR early in the bootloader, so it doesn't have the looping problem.  This also means that applications can't determine reset cause.  Unfortunately, many Nano and Mini clones don't have this update for some reason (IIRC, the Arduino boot directories never updated the .hex files?)
3) Optiboot <= v6.2:  Optiboot never had any WDT looping problem, because it uses the WDT internally, so it reset MCUSR.  But again, applications could not determine reset cause.
4) Optiboot >= v8.0 (also MCUDude version): Goes to significant effort to preserve the MCUSR content for applications, but introduces the case described in this issue.  The assumption is that if a user enables the WDT, they are "advanced" enough to also reset MCUSR themselves.  Though perhaps this needs to be documented in the WDT library or something.

Note that as of 2018, Official Arduino Nanos are using Optiboot (v4.4, I think.)
https://github.com/arduino/ArduinoCore-avr/commit/1b14cc07331268e95eddcce2cc67e29ed667e62f#diff-fce09ad94aad2ba67472cd9142f00146

kalsan, 2019-05-08:
Wow, that's awesome! Very nice to hear. Also makes things a lot clearer. Thanks for sharing!