MCUdude/MiniCore #93: 328PB pins 0, 1, 2 not working as expected

circuitsirl, 2019-05-14:
Just trying to get a basic blink sketch working.  If I do a digital or analog write to pin 1 it works as expected but also writes the same values to pins 0 and 2.  If I do a different digital write to each pin as per the code below all three pins stay high.  I do not have a separate board to test on to see if this is a hardware problem but I suppose it could be.  Pins 3, 4, and 23 all work as expected.

I have the latest Arduino AVR board installed (1.6.23) but also tried 1.6.207.  

Thank you!

```
void loop() {
  digitalWrite(redPin, HIGH);   
  delay(timeDelay);                      
  digitalWrite(redPin, LOW);   
  delay(timeDelay);  
       
  digitalWrite(grnPin, HIGH);   
  delay(timeDelay);                      
  digitalWrite(grnPin, LOW);   
  delay(timeDelay);    
  
  digitalWrite(bluPin, HIGH);   
  delay(timeDelay);                      
  digitalWrite(bluPin, LOW);   
  delay(timeDelay);                  
}
```

MCUdude, 2019-05-14:
I'm not able to look at it at the moment, but meanwhile please post the entire code and not just the content of loop().

circuitsirl, 2019-05-14:
Thanks, I have updated my first post to include the full code.

MCUdude, 2019-05-15:
I'm pretty sure there must be something wrong with your chip and/or your hardware in general. Here's a screenshot of my oscilloscope connected to digital pin 0, 1 and 2 (PB0, PB1, PB2). As you can see everything looks fine.

Are you sure you're not mixing you digital pin numbers and physical pin numbers? They're not the same.

[image]

circuitsirl, 2019-05-22:
Thank you for checking this out for me.  I am absolutely sure I am not mixing up digital/physical pin numbers.  I think it must be a hardware problem.  If I ever get another board I will try again!

closed 2019-05-22 by circuitsirl