MCUdude/MiniCore #94: MiniCore ATMEGA8 Clock Frequency Problem

tomgregg, 2019-05-16:
Hi,  I'm using the MiniCore library to test a 16MHz crystal (with caps) connected to an ATMEGA8 chip.  I simply chose the Blink example and set the desired output pin.  I'm creating HEX files and USBasp to upload the code onto the ATMEGA8.  With Board = ATMEGA8 and Clock = 16MHz External,  the LED blinks 16 times slower than expected (16.5 seconds on, 16.5 seconds off).  If I disconnect the crystal, the blink rate does not change.  I tried a few more clock settings:
12 MHz External =12.5 seconds.  And 12 MHz crystal disconnection makes no difference.
8 MHz internal = 16.5 seconds
1 MHz internal = 1.0 seconds
Any ideas whatI'm doing wrong?
Thank you

MCUdude, 2019-05-16:
The ATmega8 runs on a 1 MHz internal oscillator by default. You will have to burn the bootloader in order to set the correct fuses. Try burning the bootloader first and then upload your program using your programmer.

If you're having an issue like [this](https://github.com/MCUdude/MiniCore/issues/88) it's caused by a bug in the latest boards manager version. If you can't wait for a new release you can simply do a manual install

tomgregg, 2019-05-16:
Hi,
Thank you very much for the information, and I'm sure it will work.  Unfortunately, I started using PonyProg2000, and when I attempted to set CKSEL(0) to zero (by checking the box), I managed to brick the ATMEGA8.  Time to wait for more hardware so I can do HVSP.  But, I'm on the right track.

procsynth, 2019-05-16:
@tomgregg If you have another chip (or an Arduino) at hand you could generate a clock signal to feed in your ATMEGA8 while you set the correct fuses. I've modified the ArduinoISP.ino sketch to do just that : https://github.com/stimulu/Arduino/blob/1c9055864b51d0017464bb222b8a231f2e12405d/build/shared/examples/11.ArduinoISP/ArduinoISP/ArduinoISP.ino  
You have to uncomment line 61 and hook-up pin 3 of the Arduino to XTAL1 of your chip. Upload the sketch and try to reburn your fuses. 
I hope it helps ;)

closed 2019-11-16 by MCUdude