MCUdude/MiniCore #99: Question: how to use atmega on 1.8432mhz

nedoskiv, 2019-07-03:
I got a project that need to run on uart safe frequency, accouring to atmega datasheet 
 1.8432mhz is one of the stablest , but there is no option to run minicore on that frequency, what should I do (do not want to run on 18.432Mhz - also uart friendly)

MCUdude, 2019-07-03:
Apart from accurate delayMicroseconds, 1.8432 MHz is actually supported.

Assuming you're using an ATmega328/P you can just add these lines til boards.txt. The 1.8432MHz bootloader is already present!

```
328.menu.clock.1_8432MHz_external=1.8432 MHz external
328.menu.clock.1_8432MHz_external.upload.speed=115200
328.menu.clock.1_8432MHz_external.bootloader.low_fuses=0b1111{bootloader.cksel_bits}
328.menu.clock.1_8432MHz_external.build.f_cpu=1843200L
```

nedoskiv, 2019-07-03:
Thank you
i gonna use it on atmega8 and atmega88 but guess change "328." from your example gonna do the trick.
Should I close this issue or leave it open in case someone else have similar question?

MCUdude, 2019-07-03:
Bear in mind that the fuse settings are a little different on ATmega8 than on 88/328. Below is the correct one for ATmema8 and 88. I'm closing this issue because 1.8432 MHz, 3.6864MHz, 7.3728MHz, 11.0592MHz, and 14.7456 MHz will be available options in the next release of MiniCore.

```
8.menu.clock.1_8432MHz_external=1.8432 MHz external
8.menu.clock.1_8432MHz_external.upload.speed=115200
8.menu.clock.1_8432MHz_external.bootloader.sut_cksel_bits=111111
8.menu.clock.1_8432MHz_external.bootloader.ckopt_bit=0
8.menu.clock.1_8432MHz_external.build.f_cpu=1843200L

88.menu.clock.1_8432MHz_external=1.8432 MHz external
88.menu.clock.1_8432MHz_external.upload.speed=115200
88.menu.clock.1_8432MHz_external.bootloader.low_fuses=0b1111{bootloader.cksel_bits}
88.menu.clock.1_8432MHz_external.build.f_cpu=1843200L
```

closed 2019-07-03 by MCUdude

nedoskiv, 2019-07-29:
Hello again,
I have updated to version 2.0.3 and see 18.432 MHZ is supported, but not 1.8432MHz

MCUdude, 2019-07-29:
Well, there hasn't been a new release since either. If you need 1.8432 MHz you'll have to do a manual install for now.