MCUdude/MiniCore #100: Request Int rc. 128khz

pigglet, 2019-07-08:
Could I request for the above fuse to made available to select please, thank you for bootloader

MCUdude, 2019-07-08:
I'm not going to add a 128kHz option. Here's why:
* The clock is so low that it will have problems communicating with standard programmers (which usually runs at 250kHz.
* The oscillator is very inaccurate and isn't well suited for accurate timing
* only super slow baud rates would work
* the 1MHz option is OK for most users

Why would you need the microcontroller to run at 128kHz in the first place?

closed 2019-07-08 by MCUdude

pigglet, 2019-07-08:
Its for a device that is out in the field and needs to run for as long as
possible, as its powered by a capacitor. No interrupts are possible so
sleep is out of the question. I have achieved nano amp consumption with
this method with other mcus Thanks anyway I will find another way