MCUdude/MiniCore #102: Burning bootloader on ATMEGA8 could not find any USB device

wilson584, 2019-07-21:
I plugged in my Atmega8A into my arduino uno board. When I tried to burn the bootloader, I encountered the error 

> avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Wilson\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.3/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : stk500v2
avrdude: usbdev_open(): did not find any USB device "usb" (0x03eb:0x2104)

avrdude done.  Thank you.

Error while burning bootloader.

[image]

MCUdude, 2019-08-06:
You can't just burn the bootloader to an ATmega8 without using a dedicated programmer. Do you have an AVRISP mkii?

MCUdude, 2019-11-16:
You will have to use a dedicated programmer for this. There's plenty of tutorials around on how to do this.

closed 2019-11-16 by MCUdude