MCUdude/MiniCore #103: Support for 16.384

pcastillo87, 2019-07-24:
Hi I'm using a 16.384MHz Crustal, Can I change the boards.txt and find a bootoptiboot_flash_atmega328pb_UART1_115200_16384000L ? Thank you

MCUdude, 2019-07-25:
Why would you use a 16.384 MHz crystal? There's no real benefit as I can see. You lose accurate timing, and it's not UART friendly either. If you need a bootloader for that specific clock you're free to compile your own.

closed 2019-07-25 by MCUdude

pcastillo87, 2019-07-26:
// const uint16_t t1_comp = 62499; // for a 16MHz Crystal (16,000,000 / 256) - 1
const uint16_t t1_comp = 63999; // for a 16.384MHZ Crystal (16,384,000 / 256) - 1

Hello. Thank you for your quick answer. I hope to use an atmega328P with a crystal to build an accurate clock. I don't know if that is possible but I will try. Then when I counted the frequency with 16MHz it's less accurate that with 16.384MHz. In both cases, I used the 256 prescaler. above I leave the variables that I have been using. I appreciate your tips. Thank you